STMicroelectronics/STM32CubeH7 #159: Nucleo-STM32H743ZI2 board gets hung at SystemClock_Config();

sourabhmisal, 2021-06-11:
OS  : Ubuntu 20.04.2 LTS
IDE : STM32CubeIDE version 1.5.1

I am not able to run any program on STM32H743ZI2. I genereated code from STM32CubeIDE and flashed on it. I had added HAL_GPIO_TogglePin(LD2_GPIO_Port, LD2_Pin) with delay of 1000 ms and flashed the program. When LD2 did not blink, I tried debugging and found out that pointer stops at SystemClock_Config(); In this function, the code enters into error handler of RCC Oscillator config function
```
if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }
```
In function HAL_RCC_OscConfig located at stm32h7xx_hal_rcc.c , the code sometimes stop when waiting for HSE to get ready 
```
/* Wait till HSE is disabled */
        while(__HAL_RCC_GET_FLAG(RCC_FLAG_HSERDY) == 0U)
        {
            if((uint32_t) (HAL_GetTick() - tickstart ) > HSE_TIMEOUT_VALUE)
            {
              return HAL_TIMEOUT;
            }
        }
```

Other times, code stops running at PLL Clock 

```
/* Wait till PLL is disabled */
        while(__HAL_RCC_GET_FLAG(RCC_FLAG_PLLRDY) != 0U)
        {
          if((HAL_GetTick() - tickstart ) > PLL_TIMEOUT_VALUE)
          {
            return HAL_TIMEOUT;
          }
        }
```

STM32H7 board also heats up. Is there any way to solve this issue?

Sumatt92, 2021-06-11:
Sounds like maybe an issue with the clock configuration. IDK about that board but the h745 im using does not have an external oscillator so I use use the HSI RC for PLL source MUX (almost everything is downstream from that).

[image]

RKOUSTM, 2021-06-20:
Hi @sourabhmisal,

Thank you for your contribution. The problem seems to be related to migration to the CubeIDE. However, this public organization on GitHub is intended to promote co-development and issues posted here shall exclusively be related to the software published within the different repositories.
For any other issue (e.g. related to documentation, to hardware, to ecosystem, to user application, etc.), you shall address your questions to the [ST Community](https://community.st.com/s/topiccatalog).  There you shall look for the adequate topic and submit your question or remark.

You can use the IDE proposed by ST (IAR, MDK-ARM and SW4STM32) for example and application on STM32H7 board all of them are working well.

Please allow me to close this issue now. Thank you again for you contribution.

With regards,

closed 2021-06-20 by RKOUSTM