STMicroelectronics/STM32CubeH7 #161: Wrong paralle NOR setting

trinhnhatnam, 2021-06-24:
1. Reproduce the problem:
- Write data to parallel NOR.
- Walk through the NOR, reading each byte twice then compare. Check for mismatch. This is the code I use:

```
        int idx = 0;
        int loop = 0;
        while(1)
        {            
            uint8_t norValue1 = *((volatile uint8_t*)0x60000000 + idx);
            uint8_t norValue2 = *((volatile uint8_t*)0x60000000 + idx);
            
            if ( norValue1 != norValue2)
                printf("1: %x %x %x %x\r\n", loop, idx, norValue1, norValue2);
            
            ++idx;
            if (idx == 0xC00000)
            {
                idx = 0;
                ++loop;
            }   
        }
```

The current setting sets the read cycle exactly at 70ns. The problem is the tACC of the NOR has a maximum value of 70ns. If that happens, the FMC will not have time to sample valid data.

The fix: lengthen the cycle. In my code, I set "DataSetupTime" to 6 instead of 5. This is my code:

```
  /* Asyncronuous. Timing configuration derived from HCLK = 200MHz (5ns period). */
  Timing.AddressSetupTime      = 9;
  Timing.AddressHoldTime       = 1 /* don't care */;
  Timing.DataSetupTime         = 6;
  Timing.BusTurnAroundDuration = 4;
  Timing.CLKDivision           = 4 /* don't care */;
  Timing.DataLatency           = 2 /* don't care */;
  Timing.AccessMode            = FMC_ACCESS_MODE_B;
```

RKOUSTM, 2021-07-02:
Hi @trinhnhatnam,

Thank you for your reports.

Indeed, the `DataSetupTime` parameter defines the number of HCLK cycles to configure the duration of the data setup time. This parameter can be a value between Min_Data = 1 and Max_Data = 255,  so you can adopt it to your use case. 

This issue is not directly related to some software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to our ecosystem (namely the Cube MX tool), so please allow me to close it.

Thank you again for your reports. Do not hesitate in case you have other questions.

 We are looking forward to reading from you again.

 With regards,

trinhnhatnam, 2021-07-02:
Please don't close the bug. This bug is in the code published by ST that causes headaches for users, so please fix it. The wrong NOR settings that caused me headaches are in Drivers/BSP/STM32H747I-EVAL/stm32h747i_eval_nor.c

You are acting like insurrance companies; denying responsibility as the default policy. This is the second consecutive time you closed a bug on me without trying to discuss. Please take a look at bug #142. You guys closed it without trying to discuss. Only when I created #143, did you try to look at it.

RKOUSTM, 2021-07-02:
Hi @trinhnhatnam,

First of all, I would like to emphasize that you can reopen any closed issue if you have something else to add regarding your reported issue. 

Indeed, in order to allow a better analysis of this problem, would you please share your project that you are used for further analysis.

Thank you once more for your contribution.

With regards,

trinhnhatnam, 2021-07-02:
I already mentioned the piece of code I used to test the NOR. It's not a comprehensive test, but does the job for me. You can open an example project already in STM32H7CUBE that uses NOR, then paste my test code in somewhere to test.

RKOUSTM, 2021-08-17:
Hi @trinhnhatnam,

We tried once more to reproduce the issue you have described using the STM32H747I_EVAL BSP example published in this repository, with no success. You can used this BSP example to developed your application. I really hope you could find the root cause of this issue and, ideally, solve it. Anyway, if you ever become able to reproduce systematically the faulty behavior and find out that the problem is related to our drivers, do not hesitate to recontract me. We could reopen this issue again.

Thank you for your contribution. We are looking forward to reading from you again.

With regards,

RKOUSTM, 2021-08-17:
Hi @trinhnhatnam,

We tried once more to reproduce the issue you have described using the STM32H747I_EVAL BSP example published in this repository, with no success. You can use this BSP example to develop your application. I really hope you could find the root cause of this issue and, ideally, solve it. Anyway, if you ever become able to reproduce systematically the faulty behavior and find out that the problem is related to our drivers, do not hesitate to recontract me. We could reopen this issue again.

Thank you for your contribution. We are looking forward to reading from you again.

With regards,

closed 2021-08-17 by RKOUSTM