STMicroelectronics/STM32CubeH7 #162: stm32h7xx_ll_hsem.h / LL_HSEM_1StepLock uses HSEM_R_LOCK instead of HSEM_RLR_LOCK to while checking RLR register

usaguerrilla, 2021-07-02:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus) 

**Describe the set-up**
API Bug

**Describe the bug**

HSEM_R_LOCK seem to be the same as HSEM_RLR_LOCK but to make sure there isn't a bug later it might be a good idea to use proper Lock indication.

`
__STATIC_INLINE uint32_t LL_HSEM_1StepLock(HSEM_TypeDef *HSEMx, uint32_t Semaphore)
{
  return ((HSEMx->RLR[Semaphore] != (HSEM_R_LOCK | LL_HSEM_COREID)) ? 1UL : 0UL);
}
`

**How To Reproduce**
API bug

**Additional context**
N/A

**Screenshots**
N/A

RKOUSTM, 2021-08-06:
Hi @usaguerrilla,

Thank you for your report. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release.

Thank you once again for your contribution.

With regards,

RKOUSTM, 2021-08-06:
ST Internal Reference: 111698

RKOUSTM, 2021-12-14:
Hi @usaguerrilla,
 
I hope you are fine. The issue you reported has been fixed in the frame of version v1.9.1 of the STM32CubeH7 published recently on GitHub.

Thank you again for having reported.

With regards,