STMicroelectronics/STM32CubeH7 #163: STM32H723+CUBEH7  V1.9.0   Reset the USB Core, ResetBit is not cleared

snikeguo, 2021-07-06:
**Describe the set-up**
 * custom board(only  STM32H723ZGT6+USB3300+LM1117-3.3V+Resistor、capacitor )
 *KEIL5

**Describe the bug**
In USB_CoreReset:
Reset the USB Core, ResetBit is not cleared 

before reset:
[image]
after reset:
[image]
datasheet:
[image]
[image]

CMOD bit is 1 after reset which does not meet expectations .

**How To Reproduce**
1. Please download the IOC file：
[STCubeGenerated.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/6766791/STCubeGenerated.zip)

ALABSTM, 2022-01-30:
Hi @snikeguo, 

Thank you for this report. We will get back to you with a feedback as soon as possible. Please excuse the delay it may take us sometimes to reply. Thank you for your comprehension.

With regards,

chengyu666, 2022-03-04:
I meet the same problem on STM32H743VIT6+USB3343, I've tried almost every method online, but still not work TAT

snikeguo, 2022-03-04:
i used usb3300. pin 0 to gnd.  .everything is ok。。。

ALABSTM, 2022-04-05:
Hi @snikeguo,

Thank you again for this detailed report and the proposed workaround.

We recently published **v1.10.0** of the STM32CubeH7 firmware. Have you given it a try to check whether the issue still persists? Indeed, this latest version supports newer versions of the USB Host and Device libraries.

With regards,

chengyu666, 2022-04-06:
this fault may due to hardware error.
in my case, this problem is caused by my usb cable. its DM & DP lines are swapped, so the usb phy can't receive any signal from the host. it takes me 2days to find out the weird problem...

ALABSTM, 2022-04-12:
Hi @snikeguo and @chengyu666,

Thank you for having shared your findings. If I understand correctly, the root-cause of the issue you faced is exclusively at hardware level, mainly your custom board and your cable, not the microcontroller itself.

In this case, please allow me to close this issue. Thank you for having reported and glad to read you overcame these problems.

With regards,

closed 2024-02-12 by ALABSTM