STMicroelectronics/STM32CubeH7 #164: HAL_DSI_MspInit bad generation for STM32H747

x893, 2021-07-10:
Hi,
i use DSI Host with generation in separate file dsihost.c
In **HAL_DSI_MspInit** CubeMX and STM32IDE generate next lines

    /* DSI clock enable */
    __HAL_RCC_DSI_CLK_ENABLE();
    **( );**
    __HAL_RCC_GPIOB_CLK_ENABLE();

**error: expected expression before ')' token**

ASELSTM, 2021-07-12:
Hi @x893,

This issue is rather related to the CubeIDE tool and particularly to the CubeMx tool used for the generation of the initialization code. Actually, aspects related to tools are not handled at GitHub level. They are rather treated at **ST Community** level, precisely within the **CubeMX** dedicated [section](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) and  **CubeIDE** dedicated [section](https://community.st.com/s/topic/0TO0X000000y2j7WAA/stm32cubeide). There, you may find posts that already raised the issue you are pointing out here. Otherwise, you can create a new post to submit your issue.

As this issue is not directly related to software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to ecosystem (CubeMX tool, CubeIDE tool), please allow me to close it. Thank you for your comprehension.

With regards,

closed 2021-07-12 by ASELSTM

x893, 2021-07-12:
OK, Thanks.