STMicroelectronics/STM32CubeH7 #165: Cannot find macro definition TAMP_BKP0R_BKP

cnxiaochun, 2021-07-14:
Hi:

Team STM32

In the file stm32h7xx_ll_rtc.h

#if defined(TAMP_BKP0R_BKP)    /* This should be #if defined(TAMP_BKP0R)

__STATIC_INLINE void LL_RTC_BKP_SetRegister(TAMP_TypeDef *TAMPx, uint32_t BackupRegister, uint32_t Data)

I cannot find macro definition TAMP_BKP0R_BKP,  so "LL_RTC_BKP_SetRegister" undeclared.

Sorry to trouble you, Thanks.

RKOUSTM, 2021-07-15:
Hi @cnxiaochun, 

Thank you your contribution. Actually, the point you raised has already been dealt and fixed internally. The fix will be made available in the frame of a future release.

Thank you once more for your contribution. We are looking forward to reading from you again. 

With regards

RKOUSTM, 2021-07-15:
ST Internal Reference: 94874

RKOUSTM, 2021-07-15:
ST Internal Reference: 110701

RKOUSTM, 2021-12-14:
Hi @cnxiaochun,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.9.1 of the STM32CubeH7 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-12-14 by RKOUSTM