STMicroelectronics/STM32CubeH7 #166: Error handling in HAL_DMA_RegisterCallback

radioflash, 2021-07-14:
There are two problems with error handling in `HAL_DMA_RegisterCallback`.
Fixing these would also appease static analysis tools, even in code that is not directly affected.

## 1 - HAL_DMA_XFER_ALL_CB_ID
When `HAL_DMA_XFER_ALL_CB_ID` is passed as `CallbackID` to `HAL_DMA_RegisterCallback`, the function does not signal an error (return value of HAL_ERROR), **nor** does it do anything else (except locking and unlocking the handle).  

Expected behavior would be:
   - `pCallback` function argument is registered as handler for ALL cases (this works, is consistent with `HAL_DMA_UnRegisterCallback`, but often not really helpful)
   - The function does nothing and returns HAL_ERROR

## 2 - Invalid argument
When `CallbackID` has an invalid value (i.e. something that triggers the default branch in the switch statement), then the function should return HAL_ERROR (consistent with `HAL_DMA_UnRegisterCallback`).

I can submit a pull request if you tell me how the problem 1 behavior should be. Personally, I would slightly prefer the first alternative (i.e. registering the same callback for ALL cases), because it is least surprising and consistent with HAL_DMA_UnRegisterCallback.

ASELSTM, 2021-08-11:
ST Internal Reference: 111248

ASELSTM, 2021-08-11:
ST Internal Reference: 111247

ALABSTM, 2021-12-20:
Hi @radioflash,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.9.1** of the STM32CubeH7 published recently on GitHub. Thank you again for having reported.

With regards,

closed 2021-12-20 by ALABSTM