STMicroelectronics/STM32CubeH7 #167: Incorrect Macro IS_ADC_EXTTRIGINJEC for H7 in FW V1.9

MikeRDe, 2021-07-15:
Hi dear STM32 team,

STMH7 firmware V1.9
file: stm32h7xx_hal_adc_ex.h

The macro  `IS_ADC_EXTTRIGINJEC`  needs the additional 5 checks:

`                                         
 ((__INJTRIG__) == ADC_EXTERNALTRIGINJEC_HR1_ADCTRG2) ||  \
 ((__INJTRIG__) == ADC_EXTERNALTRIGINJEC_HR1_ADCTRG4) ||  \
 ((__INJTRIG__) == ADC_EXTERNALTRIGINJEC_LPTIM1_OUT)  ||  \
 ((__INJTRIG__) == ADC_EXTERNALTRIGINJEC_LPTIM2_OUT)  ||  \
 ((__INJTRIG__) == ADC_EXTERNALTRIGINJEC_LPTIM3_OUT)  ||  \
`

and throws an assert if  for example ADC_EXTERNALTRIGINJEC_HR1_ADCTRG2 is used as an injected trigger source parameter
in HAL_ADCEx_InjectedConfigChannel():

`  assert_param(IS_ADC_EXTTRIGINJEC(sConfigInjected->ExternalTrigInjecConv));`
 

The position in the h-file looks like as if the missing checks had been deleted accidentically. It's missing at least since V1.2.

I hope that helps,

best regards,
Michael

ASELSTM, 2021-07-16:
Hi @MikeRDe,

Thank you for this report. The issue you pointed out has been confirmed, a fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2021-07-16:
ST Internal Reference: 110734

ALABSTM, 2021-12-20:
Hi @MikeRDe,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.9.1** of the STM32CubeH7 published recently on GitHub. Thank you again for having reported.

With regards,

closed 2021-12-20 by ALABSTM

MikeRDe, 2021-12-21:
Hi ALABSTM,

I hope you are fine too. I noticed V1.9.1 and the fixed issue (and many more). Thanks for the support and have a good time!

With regards,
Michael