STMicroelectronics/STM32CubeH7 #169: `I2C_Master_Transmit` waits for `TXIS` status instead of `TXE` status

fazerlicourice7, 2021-07-23:
**Describe the set-up**
Using H745ZI on the Nuclea H745ZI-Q w/ stmcube mx v6.3.0 w/ StmCube IDE v1.6.1

**Describe the bug**
The HAL I2C library checks the `TXIS` bit in the `ISR` register even in sequential mode. This bit is used in interrupt and DMA mode, but not necessarily in sequential mode. Therefore, in sequential mode, the `HAL_I2C_Master_Transmit` function blocks until timeout.

**How To Reproduce**
1. Talk to i2c slave via hal i2c driver in a (free)rtos enabled application.

2. HAL I2C

3. Blocking / Polling mode

4. use hal i2c + freertos

ASELSTM, 2021-08-17:
Hi @fazerlicourice7,

We have tried to reproduce the issue you pointed out and unfortunately we couldn't. We have used , as you have mentioned above, the I2C in a FreeRTOS enabled application and there was no troubles with the I2C. Indeed, the communication was set as it should be and the data was successfully transmitted to the slave. We did not detect any problem with the `HAL_I2C_Master_Transmit()` function which was not blocked until the timeout.  Would you please make sure that your configuration is correct. 

Actually, the TXIS flag is set independently of the I2C transferring mode ( polling, interrupt or DMA).  Once the I2C_TXDR register becomes empty and the data to be transmitted must be written in the I2C_TXDR register an TXIS interrupt is generated. 
The TXIS will both indicates the states of the TXDR register and exactly when it should be written unlike the TXE which will indicates only the TXDR state. Thus, the check on the TXIS is more accurate. 

With regards,

ASELSTM, 2021-08-23:
Hi @fazerlicourice7,

Please allow me to close this thread. You can reopen it at any time if you have details to provide us to help you solve this issue.
Thank you for your comprehension and thank you once more for your contribution.

With regards,

closed 2021-08-23 by ASELSTM

kuro68k, 2025-02-24:
I believe I have the same issue on a G431. When using the blocking function, it checks for this flag which is never set. The interrupt code works.