STMicroelectronics/STM32CubeH7 #170: STM32H723 ADC 3 cannot be set to 12-bit resolution using LL drivers

heinwessels, 2021-08-05:
**Describe the set-up**
We are using a custom PCB with STM32H723ZG running Zephyr. We want to use ADC 3 at 12bit resolution.

**Describe the bug**
ADC3 cannot be set to 12bit resolution using the LL drivers, but always goes 10bit resolution. This is not an issue with ADC 1 or 2.

**How To Reproduce**
1. Initialize ADC 3
2. Set ADC3 resolution to 12 bit using `LL_ADC_SetResolution(adc, LL_ADC_RESOLUTION_12B);`
3. Read a high voltage (connect ADC pin to Vref+).
4. Observe the raw value returned never exceeding 1024, which shows 10bit resolution.
5. Read `ADC3_CFGR_RES` and see it's set to 10bit resolution.

**Additional context**

This is seemingly caused by a bug in the LL drivers's [`LL_ADC_SetResolution` function](https://github.com/STMicroelectronics/STM32CubeH7/blob/5975bffae9358bc2b2890a35a203d940a395efef/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h#L2921). See the code snippet below:

```
if (ADCx == ADC3)
  {
    MODIFY_REG(ADCx->CFGR, ADC3_CFGR_RES, (((Resolution & 0x10UL) | 0x08UL | (Resolution & 0x04UL))  & 0x00000018UL));
  }
```

According to the Reference Manual (DM00603761) the ADC resolution is defined in the  `ADC_CFGR` register in `RES` which is at mask `0x18`. To achieve 12bit resolution the `RES` value must be set to `00`, and for 10bit to `01`, etc. In the code snippet shown above the culprit seems to be the `| 0x08UL`, which will always set the lower bit regardless of desired `Resolution`, meaning it's impossible to set `RES` to `00`.

On the same note: `Resolution & 0x04UL` doesn't do anything since the `& 0x00000018UL` will always mask it out.

ASELSTM, 2021-08-13:
ST Internal Reference: 111930

ALABSTM, 2021-12-20:
Hi @heinwessels,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.9.1** of the STM32CubeH7 published recently on GitHub as you can see below. Thank you again for having reported.

https://github.com/STMicroelectronics/STM32CubeH7/blob/e9472e471cc8974c4a2596fc86e565241871c20e/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h#L2988-L2991

With regards,

closed 2021-12-20 by ALABSTM