STMicroelectronics/STM32CubeH7 #171: Third party FreeRTOS cmsis_os osSemaphoreWait return value

canhkha, 2021-08-16:
Wrong return in osSemaphoreWait .

The issue was detected 6 years ago.
https://community.st.com/s/question/0D50X00009XkenZ/return-value-from-cmsis-ossemaphorewait

Are you guys going to fix this?

RKOUSTM, 2021-09-03:
Hi @canhkha,

First we would like to thank you for your contribution.

The issue you pointed out is confirmed. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release. 

Thank you once again for your contribution.

With regards,

RKOUSTM, 2021-09-03:
ST Internal Reference: 112794

ALABSTM, 2022-09-06:
Hi @canhkha,

I hope you are fine. Back to you about this point you reported. According to our development teams, this point is present in CMSIS RTOS **V1** which will be **no longer supported**. In the next releases, the firmware will support **only** CMSIS RTOS **V2**. So no fix is planned.

Please allow me to close this thread in this case. Thank you for your comprehension.

Kind Regards,

closed 2022-09-06 by ALABSTM