STMicroelectronics/STM32CubeH7 #174: EEPROM emulation example bug

Irisbus, 2021-08-25:
**Describe the set-up**
The example was ported to our custom board with STM32H735V and there I discovered the bug. It is also mentioned [here](https://community.st.com/s/question/0D53W000012WR89SAG/eeprom-emulation-on-stm32h7) and [here](https://community.st.com/s/question/0D50X00009vKjpqSAC/eeprom-emulation-on-stm32h7).
IDE used is STM32CubeIDE version 1.7.0.

<br>

**Describe the bug**
I think all three examples for EEPROM emulation in this project have the same bug. On line 394 in function `EE_ReadVariable()` in eeprom.c, there should be two times the value from line 384 so it is not jumped to data.

Here should be 64:
https://github.com/STMicroelectronics/STM32CubeH7/blob/5975bffae9358bc2b2890a35a203d940a395efef/Projects/STM32H743I-EVAL/Applications/EEPROM/EEPROM_Emulation/Src/eeprom.c#L394

<br>

Here should be 64:
https://github.com/STMicroelectronics/STM32CubeH7/blob/5975bffae9358bc2b2890a35a203d940a395efef/Projects/NUCLEO-H743ZI/Applications/EEPROM/EEPROM_Emulation/Src/eeprom.c#L394

<br>

Here should be 32:
https://github.com/STMicroelectronics/STM32CubeH7/blob/5975bffae9358bc2b2890a35a203d940a395efef/Projects/NUCLEO-H7A3ZI-Q/Applications/EEPROM/EEPROM_Emulation/Src/eeprom.c#L394

<br>

**How To Reproduce**
If there is some data with value of some virtual address, the read will return incorrect data. In STM32CubeF4 examples, there it is correct (see for example [eeprom.c](https://github.com/STMicroelectronics/STM32CubeF4/blob/master/Projects/STM32446E-Nucleo/Applications/EEPROM/EEPROM_Emulation/Src/eeprom.c#L384)).

<br>

**Additional context**
N/A

<br>

**Screenshots**
N/A

ASELSTM, 2021-08-31:
ST Internal Reference: 112562

ALABSTM, 2021-12-20:
Hi @Irisbus,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.9.1** of the STM32CubeH7 published recently on GitHub as you can see below. Thank you again for having reported.

With regards, 

https://github.com/STMicroelectronics/STM32CubeH7/blob/e9472e471cc8974c4a2596fc86e565241871c20e/Projects/STM32H743I-EVAL/Applications/EEPROM/EEPROM_Emulation/Src/eeprom.c#L393

https://github.com/STMicroelectronics/STM32CubeH7/blob/e9472e471cc8974c4a2596fc86e565241871c20e/Projects/NUCLEO-H743ZI/Applications/EEPROM/EEPROM_Emulation/Src/eeprom.c#L393

https://github.com/STMicroelectronics/STM32CubeH7/blob/e9472e471cc8974c4a2596fc86e565241871c20e/Projects/NUCLEO-H7A3ZI-Q/Applications/EEPROM/EEPROM_Emulation/Src/eeprom.c#L393

closed 2021-12-20 by ALABSTM