STMicroelectronics/STM32CubeH7 #177: ADC3 Oversampling Ratio incorrectly set

heinwessels, 2021-09-16:
**The Bug**
In the function [`LL_ADC_ConfigOverSamplingRatioShift`](https://github.com/STMicroelectronics/STM32CubeH7/blob/5975bffae9358bc2b2890a35a203d940a395efef/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h#L6260-L6263) (and corresponding [`LL_ADC_GetOverSamplingRatio`](https://github.com/STMicroelectronics/STM32CubeH7/blob/5975bffae9358bc2b2890a35a203d940a395efef/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h#L6272-L6275)) the `CFGR2->OVSR` does not handle all cases correctly.

**More detail**
For example, for STM32H723's the `CFGR2->OVSR` field is different for ADC1+2 and ADC3. 

For ADC1 and ADC2:

> Bits 25:16 OSVR[9:0]: Oversampling ratio
This bitfield is set and cleared by software to define the oversampling ratio.
0: 1x (no oversampling)
1: 2x
2: 3x
...
1023: 1024x

For ADC3:

> Bits 4:2 OVSR[2:0]: Oversampling ratio
This bitfield is set and cleared by software to define the oversampling ratio.
000: 2x
001: 4x
010: 8x
011: 16x
100: 32x
101: 64x
110: 128x
111: 256x

Currently the LL dirver does not handle this special case.

**Proposed Solution**
Follow how the other functions handled this special case. For example:
```
__STATIC_INLINE void LL_ADC_ConfigOverSamplingRatioShift(ADC_TypeDef *ADCx, uint32_t Ratio, uint32_t Shift)
{
  MODIFY_REG(ADCx->CFGR2, ADC_CFGR2_OVSS, Shift);
#if defined(ADC_VER_V5_V90)
  if (ADCx == ADC3)
  {
    MODIFY_REG(adc->CFGR2, ADC3_CFGR2_OVSR, (Ratio - 1UL) << ADC3_CFGR2_OVSR_Pos);
  }
  else {
#endif   /* ADC_VER_V5_V90 */
  MODIFY_REG(adc->CFGR2, ADC_CFGR2_OVSR, (Ratio - 1UL) << ADC_CFGR2_OVSR_Pos);
#if defined(ADC_VER_V5_V90)
  }
#endif   /* ADC_VER_V5_V90 */
}
```

FRASTM, 2021-09-17:
Could this issue be rasied by the https://github.com/STMicroelectronics/STM32CubeH7/issues/170?

heinwessels, 2021-09-17:
> Could this issue be a duplicate of the #170?

@FRASTM definitely not, since I raised both these issues myself. They are probably related, not duplicates. They point to different functions:

[#170](https://github.com/STMicroelectronics/STM32CubeH7/issues/170): `LL_ADC_SetResolution(adc, LL_ADC_RESOLUTION_12B);`
#177: `LL_ADC_ConfigOverSamplingRatioShift(...)`

FRASTM, 2021-09-17:
internal bug https://intbugzilla.st.com/show_bug.cgi?id=113809

closed 2023-04-10 by ALABSTM