STMicroelectronics/STM32CubeH7 #187: Add support for ADC LDO output voltage ready bit

fivdi, 2021-10-31:
Revision V STM32H743 devices have an "ADC LDO output voltage ready bit" in the "ADC interrupt and status register". In the [reference manual](https://www.st.com/resource/en/reference_manual/rm0433-stm32h742-stm32h743753-and-stm32h750-value-line-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) this bit is called LDORDY and it is bit 12 in the ADC_ISR register. 

STM32CubeH7 V1.9.0 doesn't provide support for the LDORDY bit. For example, there are no defines for LDORDY in [stm32h743xx.h](https://raw.githubusercontent.com/STMicroelectronics/STM32CubeH7/v1.9.0/Drivers/CMSIS/Device/ST/STM32H7xx/Include/stm32h743xx.h) and there is no low level functionality for accessing LDORDY in [stm32h7xx_ll_adc.h](https://github.com/STMicroelectronics/STM32CubeH7/blob/v1.9.0/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h)

It would be good if support for the LDORDY bit in the ADC_ISR register was added to STM32CubeH7.

Initial tests with functions that I implemented myself for accessing the LDORDY bit in the ADC_ISR register indicate that the bit functions as expected with a [WeAct Studio STM32H743 Core Board](https://github.com/WeActTC/MiniSTM32H7xx).

The same issue may exist for other STM32H7 devices.

RKOUSTM, 2021-11-15:
Hi @fivdi,

Thank you for your contribution. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release.

Thank you once again for your contribution.

With regards,

RKOUSTM, 2021-11-15:
ST Internal Reference: 117529.

closed 2023-04-10 by ALABSTM

fivdi, 2023-04-10:
Thank you.

For anyone interested, support for the LDORDY bit in the ADC_ISR register was added to stm32h7xx_ll_adc.h with [ADC_ISR_LDORDY](https://github.com/STMicroelectronics/STM32CubeH7/blob/v1.11.0/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h#L648) and [LL_ADC_IsActiveFlag_LDORDY(ADC_TypeDef *ADCx)](https://github.com/STMicroelectronics/STM32CubeH7/blob/v1.11.0/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h#L7523-L7532).

[stm32h743xx.h](https://raw.githubusercontent.com/STMicroelectronics/STM32CubeH7/v1.11.0/Drivers/CMSIS/Device/ST/STM32H7xx/Include/stm32h743xx.h) was also appropriately modified.