STMicroelectronics/STM32CubeH7 #190: [STM32H750-DB] ExtMem_Application: hardfault caused and Unable to jump to Application

EmbeddedAyoub, 2021-11-09:
Hello,

I am using the STM32H750-DB and I am trying to copy a code from External Flash to External SDRAM to run it. I have followed what is described in the Readme.txt for :

- BootROM : Execution From External SDRAM , Data on Internal SRAM

I can easily run a code from external Flash, but not from external SDRAM.

Here is my Linkerscript Memory for extrenal Application:

```
 MEMORY
{
FLASH (rx)      : ORIGIN = 0xD0000000, LENGTH = 32768K
RAM (xrw)      : ORIGIN = 0x20000000, LENGTH = 128K
ITCMRAM (xrw)      : ORIGIN = 0x00000000, LENGTH = 64K
}
```
also I have well set the Memories in the "Memory.h" file :

`#define DATA_AREA           USE_INTERNAL_SRAM
#define CODE_AREA           USE_EXTERNAL_SDRAM
#define BINARY_AREA         USE_SPI_NOR
`
Also the Interrupt vector :

EmbeddedAyoub, 2021-11-10:
Hello,

Issue resolved! 

The SDRAM is by default set as XN (Execute Not Permitted) so you need to configure the MPU and enable the Execution for the external SDRAM.

Ayoub

closed 2021-11-10 by EmbeddedAyoub

ALABSTM, 2021-11-15:
Hi @EmbeddedAyoub,

Glad to read you found the issue and overcame it. Users facing the same problem would find your post very helpful. Thank you.

With regards,

EmbeddedAyoub, 2021-11-15:
> Hi @EmbeddedAyoub,
> 
> Glad to read you found the issue and overcame it. Users facing the same problem would find your post very helpful. Thank you.
> 
> With regards,

Welcome :)