STMicroelectronics/STM32CubeH7 #191: ADC clock frequency greater than manimum allowed in STM32H723ZG example

fivdi, 2021-11-14:
**Issue 1**

Table 79 on page 156 of the [STM32H723ZG Datasheet](https://www.st.com/resource/en/datasheet/stm32h723zg.pdf) describes the 16-bit ADC characteristics of the STM32H723ZG. Table 79 states that the maximum ADC clock frequency is 50 MHz. In addition, if the ADC clock frequency is 50 MHz, Table 79 also states that BOOST must be set to 11 (binary).

On the other hand, the NUCLEO-H723ZG [ADC_AnalogWatchdog eample](https://github.com/STMicroelectronics/STM32CubeH7/tree/v1.9.0/Projects/NUCLEO-H723ZG/Examples_LL/ADC/ADC_AnalogWatchdog) sets the ADC clock frequency to 65 MHz [here](https://github.com/STMicroelectronics/STM32CubeH7/blob/v1.9.0/Projects/NUCLEO-H723ZG/Examples_LL/ADC/ADC_AnalogWatchdog/Src/main.c#L250) (520 MHz / 4 / 2 = 65 MHz) which is above the allowed maximum of 50 MHz and never sets BOOST to 11.

**Issue 2**

If I understand the datasheet and reference manual correctly, [this code](https://github.com/STMicroelectronics/STM32CubeH7/blob/v1.9.0/Projects/NUCLEO-H723ZG/Examples_LL/ADC/ADC_AnalogWatchdog/Src/main.c#L495-L498) in function `Activate_ADC`:

```c
    /* Run ADC self calibration */
    // todo : define which calib parameter has to be used
    // LL_ADC_StartCalibration(ADC1, LL_ADC_CALIB_OFFSET, LL_ADC_SINGLE_ENDED);
    // LL_ADC_StartCalibration(ADC1, LL_ADC_CALIB_OFFSET_LINEARITY, LL_ADC_SINGLE_ENDED);
```

should be:

```c
    /* Run ADC self calibration */
    LL_ADC_StartCalibration(ADC1, LL_ADC_CALIB_OFFSET_LINEARITY, LL_ADC_SINGLE_ENDED);
```

The way the code is currently structured, `Activate_ADC` is only called once form `main` which implies that `LL_ADC_StartCalibration(ADC1, LL_ADC_CALIB_OFFSET_LINEARITY, LL_ADC_SINGLE_ENDED)` should be called.

ALABSTM, 2022-01-30:
Hi @fivdi,

Thank you for this report. We will get back to you with a feedback as soon as possible. Please excuse the delay it may take us sometimes to reply. Thank you for your comprehension.

With regards,

ASELSTM, 2023-04-07:
Hi @fivdi,

Referring to the ADC Clock scheme within reference manual [RM0468](https://www.st.com/resource/en/reference_manual/rm0468-stm32h723733-stm32h725735-and-stm32h730-value-line-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) Figure 161,   a clock divider factor of 2 is applied by default to the clock provided to the ADC analog block `(fadc_ker_ck)` .

Furthermore, it is up to the user to choose whether to perform the ADC calibration or not and it should be done only during the ADC hardware calibration. There is thus no needs to execute this code section each time.

Hope this would help you. Please allow me thus to close this thread.

With regards,

closed 2023-04-07 by ASELSTM

fivdi, 2023-09-03:
@ASELSTM yes, that helps, it's clearer now, thank you.