STMicroelectronics/STM32CubeH7 #194: .pdata in freeRTOS MPU linker script is never used

Viatorus, 2021-11-19:
Since the section .pbss already includes all `privileged_data`, the .pdata is obsolete.

https://github.com/STMicroelectronics/STM32CubeH7/blob/ccb11556044540590ca6e45056e6b65cdca2deb2/Projects/NUCLEO-H743ZI/Applications/FreeRTOS/FreeRTOS_MPU/SW4STM32/STM32H743ZI_Nucleo/STM32H743ZITx_FLASH.ld#L157-L178

Is this intended, a left over or a bug?

ALABSTM, 2022-01-30:
Hi @Viatorus,

Thank you for this report. We will get back to you with a feedback as soon as possible. Please excuse the delay it may take us sometimes to reply. Thank you for your comprehension.

With regards,

RKOUSTM, 2022-02-25:
Hi @Viatorus,

Thank you for this report. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release.

Thank you once more for your report.

With regards

With regards,

RKOUSTM, 2022-02-25:
ST Internal Reference: 123529

ALABSTM, 2023-04-10:
Hi @Viatorus,

I hope you are doing well. Our development teams see no harm to keep the `.pdata` section in the linker file. Besides, as they pointed out, this section is used for initialization.

Please allow me to close this issue. Thank you for your comprehension.

With regards,

closed 2023-04-10 by ALABSTM