STMicroelectronics/STM32CubeH7 #201: BSP driver for STM32H743I/753I -EVAL, fix for using LEDs without the I2C I/O extender

pavel-a, 2022-01-29:
This patch fixes LEDs directly connected to GPIOs when the I2C I/O extender is disabled.
File:   Drivers/BSP/STM32H743I-EVAL/stm32h743i_eval.c

https://github.com/STMicroelectronics/STM32CubeH7/blob/e9472e471cc8974c4a2596fc86e565241871c20e/Drivers/BSP/STM32H743I-EVAL/stm32h743i_eval.c#L99-L108
**Describe the set-up**
 * Board: STM32H743I/753I EVAL-2
 * IDE: any (I used CubeIDE)

**Describe the bug**
I needed to reuse the I2C for the I/O extender for other purpose, so removed `#define USE_BSP_IO_CLASS `
Then two LEDs connected directly to MCU (red and green) stopped working.  
Fix proposed below.  
Maybe this applies to other similar boards.

**How To Reproduce**
see above

**Additional context**

The fragment above should be:

```
static const  uint32_t LED_PIN[LEDn] = {LED1_PIN,
#if (USE_BSP_IO_CLASS > 0)
                                        LED2_PIN,
#else
                                       0,
#endif
                                        LED3_PIN,
#if (USE_BSP_IO_CLASS > 0)
                                        LED4_PIN
#else
                                       0,
#endif
                                       };

```

Patch below:

```

diff --git a/Drivers/BSP/STM32H743I-EVAL/stm32h743i_eval.c b/Drivers/BSP/STM32H743I-EVAL/stm32h743i_eval.c
index eb5a8c9b..6628ba70 100644
--- a/Drivers/BSP/STM32H743I-EVAL/stm32h743i_eval.c
+++ b/Drivers/BSP/STM32H743I-EVAL/stm32h743i_eval.c
@@ -99,10 +99,14 @@ static GPIO_TypeDef* LED_PORT[LEDn] = {
 static const  uint32_t LED_PIN[LEDn] = {LED1_PIN,
 #if (USE_BSP_IO_CLASS > 0)
                                         LED2_PIN,
+#else
+                                        0,
 #endif
                                         LED3_PIN,
 #if (USE_BSP_IO_CLASS > 0)
                                         LED4_PIN
+#else
+                                        0,
 #endif
                                        };
 
-- 
```

RKOUSTM, 2022-04-11:
Hi @pavel-a,

Thank you for your report. An internal tracker has been logged and a fix will be implemented and made available in the frame of a future release.

Thank you again for you report.

With regards,

RKOUSTM, 2022-04-11:
ST Internal Reference: 126180

closed 2024-03-14 by ALABSTM