STMicroelectronics/STM32CubeH7 #203: ptp problem

CoderHSF, 2022-02-28:
**  Problem **
STM32H750XBH6 performs synchronization with other PTP master devices on the network through LAN8720A using PTP timestamp. H7 acts as Slave and other devices act as Master, and I have set the TSVER2ENA register to enable. When Master sends the synchronization data packet of PTP V2 version, the data received by H7 is normal; When Master sends PTP V1 data packets, the MAC layer of H7 stops receiving all data (including non PTP data packets), which is the case every time it is started. The HAL Drivers Verision I used is V1.6.0 / 28-June-2019. Do I have any incorrect settings? Or is this a bug in the H7 chip series? Is there a solution?

ASELSTM, 2022-03-23:
Hi @CoderHSF,

We just made available a fully reworked ETH HAL driver via pull-request [stm32h7xx_hal_driver#16](https://github.com/STMicroelectronics/stm32h7xx_hal_driver/pull/16).  Would you please try this reworked driver and check if you still have the same problem. 

Please note also that this new ETH HAL driver **breaks the compatibility** with the previous one. We are looking forward to reading your feedback.

With regards,

ASELSTM, 2022-09-30:
Hi @CoderHSF,

Please allow me to close this thread as the new ETH driver is now available on the STM32CubeH7 firmware package.

With regards,

closed 2022-09-30 by ASELSTM