STMicroelectronics/STM32CubeH7 #205: Unclear software licensing

sebhub, 2022-03-11:
For example, the file [Drivers/CMSIS/Device/ST/STM32H7xx/Include/stm32h723xx.h](Drivers/CMSIS/Device/ST/STM32H7xx/Include/stm32h723xx.h) contains this license header:
```c
  * Copyright (c) 2019 STMicroelectronics.
  * All rights reserved.
  *
  * This software is licensed under terms that can be found in the LICENSE file
  * in the root directory of this software component.
  * If no LICENSE file comes with this software, it is provided AS-IS.
```
However, there is no `LICENSE` file in the repository. The closest license file is [Drivers/CMSIS/Device/ST/STM32H7xx/License.md](Drivers/CMSIS/Device/ST/STM32H7xx/License.md) which states that this file is under BSD-3-Clause license. This contradicts the top level license overview in [LICENSE.md](LICENSE.md) which states:
```
| Component                       | Copyright                                             | License                                      |
|:------------------------------- |:----------------------------------------------------- |:-------------------------------------------- |
| CMSIS                           | ARM Limited                                           | Apache License 2.0                           |
| CMSIS Device                    | ARM Limited - STMicroelectronics                      | Apache License 2.0                           |
```
Could you please clarify this.

In addition, this software is intended to be used by applications using STMicroelectronics chips. It may be integrated in project-specific repositories. In order to manage open source software licenses, the software industry established the [Software Data Package Exchange](https://spdx.dev) standard which is an international open standard (ISO/IEC 5962:2021). It would be really nice if STMicroelectronics would support this and add SPDX license identifiers to the source files.

ALABSTM, 2022-04-13:
ST Internal Reference: 126404

ALABSTM, 2022-06-02:
Fixed in commit 0c3a06c18bd2a87b006a84f997f2b0b56d97a113

closed 2022-06-02 by ALABSTM

sebhub, 2022-06-02:
Actually, this is not what I intended. Why did you change the license of this code from BSD-3-Clause to Apache License 2.0? The Apache License 2.0 has an attribution clause which makes it difficult to handle if you include the code in third-party projects.

ALABSTM, 2022-06-02:
Hi @sebhub,

Actually, our CMSIS Device components are licensed under Apache 2.0 terms. I simply updated the content of the file to be compliant with the intended licensing.

With regards,

sebhub, 2022-06-02:
Ok, this is a step backwards. Please consider to license your own code under BSD-3-Clause.