STMicroelectronics/STM32CubeH7 #208: BSP USART1 pins should be  PA9 and PA10 for STM32H747I-DISCO

VictorTagayun, 2022-03-20:
As per readme,  USART1 pins are PB14 and PB15.
https://github.com/STMicroelectronics/STM32CubeH7/blob/b340b13929e36a3427b8d94e8b1006022f82273f/Projects/STM32H747I-DISCO/Examples/UART/UART_WakeUpFromStopUsingFIFO/readme.txt#L47

Pls chk if it should be.
PA9  = USART1_TX  
PA10 = USART1_RX

ASELSTM, 2022-04-04:
Hi @VictorTagayun,

You are right the UART Tx and Rx pins mentioned in the readme.txt file are uncoherent with the pins actually configured. The reademe.txt file should then be updated as follow: 

```diff
Board: STM32H747I-DISCO
-Tx Pin: PB.14
+Tx Pin: PA.9
-Rx Pin: PB.15
+Rx Pin: PA.10
```

This fix will be implemented and made available in the frame of a next release.

With regards,

ASELSTM, 2022-04-04:
ST Internal Reference: 125777

ASELSTM, 2023-04-06:
Hi @VictorTagayun,

Thank you for your contribution. This issue has been fixed in the frame of version v1.11.0 of the STM32CubeH7. Please allow me then to close this thread.

With regards,

closed 2023-04-06 by ASELSTM