STMicroelectronics/STM32CubeH7 #210: implicit declaration LL_RCC_CalcPLLClockFreq

daniel-starke, 2022-03-24:
**Describe the set-up**
 * Compile framework with PlatformIO and compiler flag `-Wimplicit-function-declaration` (does not define `USE_FULL_LL_DRIVER`).

**Describe the bug**
`stm32h7xx_ll_utils.c` uses `LL_RCC_CalcPLLClockFreq()` in several places. `stm32h7xx_ll_rcc.h` is included, which declares the function. However, the declaration of this function is gates by the definition of `USE_FULL_LL_DRIVER`. Failing to define `USE_FULL_LL_DRIVER` results in a compiler warning "implicit declaration of function 'LL_RCC_CalcPLLClockFreq'". This may result in linker issues.

**How To Reproduce**
Compile `stm32h7xx_ll_utils.c` with the compiler flag `-Wimplicit-function-declaration` and without defining `USE_FULL_LL_DRIVER`.

**Additional context**
I propose to move the declaration of `LL_RCC_CalcPLLClockFreq()` outside of the `USE_FULL_LL_DRIVER` block in `stm32h7xx_ll_rcc.h`.

HBOSTM, 2023-09-07:
Hello @daniel-starke,

First of all thank you for this contribution. You cannot use this driver without the USE_FULL_LL_DRIVER definition. 

Best Regards,
Houssine

daniel-starke, 2023-09-07:
I know. The point is that `USE_FULL_LL_DRIVER` does not gate all places in a consistent manner.

KRASTM, 2024-01-29:
ST Internal Reference: 171944

TOUNSTM, 2024-03-21:
Fixed in abbf9ca1b36c9abdb9339e86f0b7b2f4e79edc35

closed 2024-03-21 by TOUNSTM