STMicroelectronics/STM32CubeH7 #214: HAL_MMC_ReadBlocks CRC error in 8 bit mode SDMMC

Judicaelkouadio, 2022-04-15:
Hi,

I'm working on SDMMC on STM32U5 nucleo board.

Everything works fine for 1 and 4 bit mode.

But, in 8 bit mode I can write data with HAL_MMC_WriteBlocks but when I try to read back with HAL_MMC_ReadBlocks, it always gets SDMMC_FLAG_DCRCFAIL.

 

Have you ever met the same error? If yes, how did you solve it.

 

Thanks in advance

ASELSTM, 2022-04-19:
Hi @Judicaelkouadio,

You have mentioned in your comments that you are working on **STM32U5 nucleo board**. However, you have reported the issue on the [STM32CubeH7](https://github.com/STMicroelectronics/STM32CubeH7) repository. Would you please clarify whether the issue is detected on the STM32U5Cube or SMT32H7Cube package ? 

With regards,

Judicaelkouadio, 2022-04-19:
Hi @ASELSTM,

I use STM32U5Cube.
I didn't find the STM32CubeU5 repository. This is why I posted it here.

But the issue is for STM32U5.

Any idea?

Best regards.

ASELSTM, 2023-04-07:
Hi @Judicaelkouadio,

Would you please try to check with the latest version of the [STM32CubeU5](https://github.com/STMicroelectronics/STM32CubeU5) and check if you are still having the same issue.

With regards,