STMicroelectronics/STM32CubeH7 #217: QSPI FIFO Threshold register does not accept values greater than 15

jealcuna, 2022-05-05:
**Describe the set-up**
 * STM32H750VB mcu

**Describe the bug**
QUAD SPI has a fifo threshold register that can accept values from 0 to 31. It has 5 bits size. The 5 bit can not be modified making a possible configuration to 16 bytes 0xf.

**How To Reproduce**
1. Follow the steps to reproduce the bug here https://github.com/STMicroelectronics/STM32CubeH7/issues/216#issue-1226165798

2. Modify the mask to 0x1f

3. Try to set fifo threshold with values greater than 15, and 5 bits always keep 0.

ASELSTM, 2022-06-07:
Hi @jealcuna,

Please allow me to close this thread as it is a duplicate of #216 .

With regards,

closed 2022-06-07 by ASELSTM

jealcuna, 2022-06-07:
> Hi @jealcuna,
> 
> Please allow me to close this thread as it is a duplicate of #216 .
> 
> With regards,

But the issue persist even when mask was fixed. So I am not sure if both issues are related.