STMicroelectronics/STM32CubeH7 #224: Ethernet transmit is hang on osSemaphoreAcquire( TxPktSemaphore ) inside low_level_output()

ktrofimov, 2022-06-20:
Not really a driver issue, but it should work with ethernetif.c together...
ethernetif.c is taken from STM32H7 example (https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Projects/STM32H743I-EVAL/Applications/LwIP/LwIP_HTTP_Server_Netconn_RTOS/Src/ethernetif.c)
With 10 ICMP packets + 10 TCP connections per second Ethernet hangs after some time (could be any time from 5 minutes to 15 hours)

Ethernet transmit hangs forever on `while( osSemaphoreAcquire( TxPktSemaphore, TIME_WAITING_FOR_INPUT ) != osOK )`:

```c
static err_t low_level_output(struct netif *netif, struct pbuf *p)
{
 . . .
 if( HAL_ETH_Transmit_IT(&heth, &TxConfig ) != HAL_OK )
 {
    printf( "HAL ETH Tx IT Error\n" );
 }
 while( osSemaphoreAcquire( TxPktSemaphore, TIME_WAITING_FOR_INPUT ) != osOK )
    ;
}
```
This semaphore supposed to be cleared either in`HAL_ETH_TxCpltCallback()`:
```c
void HAL_ETH_TxCpltCallback(ETH_HandleTypeDef *heth)
{
  osSemaphoreRelease(TxPktSemaphore);
}
```

ktrofimov, 2022-06-23:
Possible solutions (has to be tested) - add code to release `TxPktSemaphore` in case of DMA Tx error:
```c
void HAL_ETH_ErrorCallback(ETH_HandleTypeDef *heth)
{
  if((HAL_ETH_GetDMAError(heth) & ETH_DMACSR_RBU) == ETH_DMACSR_RBU)
  {
	// ETH DMA Rx Error
    osSemaphoreRelease(RxPktSemaphore);
  }

+  if((HAL_ETH_GetDMAError(heth) & ETH_DMACSR_TBU) == ETH_DMACSR_TBU)
+  {
+	// ETH DMA Tx Error\n" );
+    osSemaphoreRelease(TxPktSemaphore);
+  }

}
```

ASELSTM, 2022-06-24:
ST Internal Reference: 130930

ktrofimov, 2022-07-29:
Just stumbled upon the same issue again. Now in void ethernet_link_thread(void* argument)
Steps to reproduce:
1) pull cable out
2) disconnect or disable DHCP server
3) plug cable in
4) pull cable out again
5) plug it in again
6) connect or enable DHCP server
somewhere in this sequence ethernet_link_thread hangs on xQueueSemaphoreTake():
[image]
Notice the difference in ST code (HAL_ETH_Start() instead of HAL_ETH_Start_IT() ):

```c
if(linkchanged)
{
      /* Get MAC Config MAC */
      HAL_ETH_GetMACConfig(&heth, &MACConf);
      MACConf.DuplexMode = duplex;
      MACConf.Speed = speed;
      HAL_ETH_SetMACConfig(&heth, &MACConf);
      HAL_ETH_Start(&heth);
      netif_set_up(netif);
      netif_set_link_up(netif);
}
```
but HAL_ETH_Stop_IT() : 
```c
if(netif_is_link_up(netif) && (PHYLinkState <= LAN8742_STATUS_LINK_DOWN))
{
    HAL_ETH_Stop_IT(&heth);
    netif_set_down(netif);
    netif_set_link_down(netif);
}
```
while my code have  xxx_IT in both places (I feel this is a correct way), but it doesn't help.

ktrofimov, 2022-07-29:
It looks like that proposed solution (https://github.com/STMicroelectronics/STM32CubeH7/issues/224#issuecomment-1163875733) works.

ktrofimov, 2022-07-29:
Most interesting, that despite of that hung, DHCP client is able to receive address and the rest firmware works as supposed to be.

fishjimi, 2022-08-13:
I had the same problem today. 
After debugging, I also found the problem here.
This can be solved simply
``` c
if(linkchanged)
{
  /* Get MAC Config MAC */
  HAL_ETH_GetMACConfig(&heth, &MACConf);
  MACConf.DuplexMode = duplex;
  MACConf.Speed = speed;
  HAL_ETH_SetMACConfig(&heth, &MACConf);
  //HAL_ETH_Start(&heth);
  HAL_ETH_Start_IT(&heth);  //change to xxx_IT
  netif_set_up(netif);
  netif_set_link_up(netif);
}
```

a1292999652, 2022-09-21:
The same problem still exists today, when will it be fixed?
[image]
[image]

a1292999652, 2022-09-21:
> 我今天遇到了同样的问题。 经过调试，我也发现了这里的问题。 这个可以简单解决
> 
> ```c
> if(linkchanged)
> {
>   /* Get MAC Config MAC */
>   HAL_ETH_GetMACConfig(&heth, &MACConf);
>   MACConf.DuplexMode = duplex;
>   MACConf.Speed = speed;
>   HAL_ETH_SetMACConfig(&heth, &MACConf);
>   //HAL_ETH_Start(&heth);
>   HAL_ETH_Start_IT(&heth);  //change to xxx_IT
>   netif_set_up(netif);
>   netif_set_link_up(netif);
> }
> ```

But every time the code is generated, it goes back to the original

pavel-a, 2022-10-05:
As [suggested on the forum](https://community.st.com/s/question/0D53W00001oImr4SAC/bug-stm32-lwip-ethernet-driver-tx-deadlock), the reason is incorrect use of LwIP API (or use of obsolete examples).

The affected examples under [Projects/](https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Projects)  should be fixed.

ASELSTM, 2023-04-07:
Hi @ktrofimov,

Thank you for your contribution. This issue has been fixed in the frame of version v1.11.0 of the STM32CubeH7. Please allow me then to close this thread.

With regards,

closed 2023-04-07 by ASELSTM