STMicroelectronics/STM32CubeH7 #231: ethernetif.c issues

ra1u, 2022-08-17:
We have detected several issues in [ethernetif.c](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Projects/NUCLEO-H743ZI/Applications/LwIP/LwIP_HTTP_Server_Netconn_RTOS/Src/ethernetif.c)
For refernce we used this on NUCLEO-H743ZI, but bug likley applies also on on related boards as most of this code is shared.

1. Race condition in thread responsible for [ifup](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Projects/NUCLEO-H743ZI/Applications/LwIP/LwIP_HTTP_Server_Netconn_RTOS/Src/ethernetif.c#L674-L675) and [ifdown](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Projects/NUCLEO-H743ZI/Applications/LwIP/LwIP_HTTP_Server_Netconn_RTOS/Src/ethernetif.c#L635-L636)  . Lwip requres serialised acess when calling [`netif_set_up`](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Middlewares/Third_Party/LwIP/src/core/netif.c#L846) and [`netif_set_up`](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Middlewares/Third_Party/LwIP/src/core/netif.c#L990)

2. Race condition in memory allocation. As function [`HAL_ETH_RxAllocateCallback`](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Projects/NUCLEO-H743ZI/Applications/LwIP/LwIP_HTTP_Server_Netconn_RTOS/Src/ethernetif.c#L685) runs in interrupt, it is not synchronized with related [`LWIP_MEMPOOL_FREE`](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Projects/NUCLEO-H743ZI/Applications/LwIP/LwIP_HTTP_Server_Netconn_RTOS/Src/ethernetif.c#L425) . We observed, that [`LWIP_MEMPOOL_ALLOC`](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Projects/NUCLEO-H743ZI/Applications/LwIP/LwIP_HTTP_Server_Netconn_RTOS/Src/ethernetif.c#L685)  can return memory that was not yet freed, but there might be also other effects of undefined behavior.

3. There is not clear how synchronization between code running from threads and interrupts is synchronized. It can happen that code in [`ethernetif_input`](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Projects/NUCLEO-H743ZI/Applications/LwIP/LwIP_HTTP_Server_Netconn_RTOS/Src/ethernetif.c#L359) thread, is interrupted by other thread that [sends data](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Projects/NUCLEO-H743ZI/Applications/LwIP/LwIP_HTTP_Server_Netconn_RTOS/Src/ethernetif.c#L31) followed by interrupt from ISR. There is at least race condition over [eth->gState](https://github.com/ra1u/STM32CubeH7/blob/master/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_eth.c#L1054), because such requirement upon function entry can not be satisfied trough function execution.

All this issues results in undefined behavior and unreliable operation.

Looking forward to move this issue forward and find reliable solutions.

Kind regards, Luka

ASELSTM, 2023-07-12:
ST Internal Reference: 157151

ASEHSTM, 2024-03-14:
Hello @ra1u, 

Thank you for your report.
1. In regard to the first point, you are correct, we need to ensure thread-safe access when calling the `netif_xxx()` functions. It will be fixed soon.

2. For the second point, we want to clarify that `HAL_ETH_RxAllocateCallback()` is not actually called from an interrupt handler. Instead, the interrupt handler calls `RX_Callback()` which releases a semaphore to inform the thread that a packet is ready for processing. The function is then called to allocate memory for the packet. When using `FreeRTOS` (NO_SYS = 0), calls for memory allocation are protected using a mutex. The LWIP architecture also handles serialization of Read/Write requests when running in OS context, and the Sequential-style APIs use RTOS facilities to exchange messages with the TCPIP thread.

3. Finally, in regard to the third point, we want to clarify that `HAL_ETH_Read()` and `HAL_ETH_Transmit()` do not change the `gState` field. They only query the field to check that the ETH peripheral has started. The serialization of Read/Write requests is handled by the LWIP architecture when running in OS context.

We hope this explanation helps clarify the issues you reported. 
With Regards,

ASEHSTM, 2024-03-27:
Hello @ra1u,

Could you please check if the first issue has been resolved in this [version](https://github.com/STMicroelectronics/STM32CubeH7/tree/master)?

With Regards,

ASEHSTM, 2024-04-18:
Hello @ra1u,

Please allow me to close this issue. You can reopen it at any time if you have any details to share with us.

With Regards,

closed 2024-04-18 by ASEHSTM