STMicroelectronics/STM32CubeH7 #235: Linker script issue

smdjeff, 2022-10-01:
https://github.com/STMicroelectronics/STM32CubeH7/blob/4cb7c570fa5f2a362193d713a70b43ab02c68197/Projects/STM32H735G-DK/Applications/LwIP/LwIP_TCP_Echo_Server/STM32CubeIDE/STM32H735IGKX_FLASH.ld#L41

Pretty sure there's only 32k here not 288k.
Also, you're leaving about 320k on the table at 0x24000000

smdjeff, 2022-10-01:
Complete and correct linker scripts for all parts would be super handy to developers.

ASELSTM, 2023-05-22:
ST Internal Reference: 152716

ALABSTM, 2024-01-24:
Hi @smdjeff,

I hope you are fine. Please excuse the time taken to get back to you. The fix has just been uploaded. Thank you again for having reported.

With regards,

ALABSTM, 2024-01-24:
Fixed in commit c8c46c579b90011af4bf5e0c18ccf8a324c8e01c

closed 2024-01-24 by ALABSTM