STMicroelectronics/STM32CubeH7 #241: STM32H7B3I-DK FatFs / uSD / DMA

fdelgehier, 2022-10-19:
**FATFS/µSD/DMA** request.

In the **STM32H7B3I-DK** example, FatFs/FatFs_uSD_DMA_Standalone/FatFs/App/app_fatfs.c
The code is an example to make FATFS/μSD/DMA modules work,
but I have the impression that the example uses its 'polling' functions.
There are no 'Callback' functions that allow you to indicate the end of a write or a reading on a µSD.

Can you confirm if this is a mistake ?
Thank you for your attention

https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Projects/STM32H7B3I-DK/Applications/FatFs/FatFs_uSD_DMA_Standalone/FatFs/App/app_fatfs.c

HBOSTM, 2023-02-10:
Hello @fdelgehier,
Thank you for your contribution.
I don't agree with you on this point, firstly we are using BSP_SD_Write/ReadBlocks_DMA int the SD_write/Read  function, https://github.com/STMicroelectronics/STM32CubeH7/blob/43c9e552ba1c038577c48723d96ca8c825b11987/Projects/STM32H7B3I-DK/Applications/FatFs/FatFs_uSD_DMA_Standalone/FatFs/Target/sd_diskio.c#L146   
https://github.com/STMicroelectronics/STM32CubeH7/blob/43c9e552ba1c038577c48723d96ca8c825b11987/Projects/STM32H7B3I-DK/Applications/FatFs/FatFs_uSD_DMA_Standalone/FatFs/Target/sd_diskio.c#L203
 Secondary there are the write and read Transfer completed callbacks.

HBOSTM, 2023-02-10:
Hello @fdelgehier,

Thank you for your contribution.
I don't agree with you on this point, firstly we use BSP_SD_Write/ReadBlocks_DMA int the SD_write/Read  function, https://github.com/STMicroelectronics/STM32CubeH7/blob/43c9e552ba1c038577c48723d96ca8c825b11987/Projects/STM32H7B3I-DK/Applications/FatFs/FatFs_uSD_DMA_Standalone/FatFs/Target/sd_diskio.c#L146   
https://github.com/STMicroelectronics/STM32CubeH7/blob/43c9e552ba1c038577c48723d96ca8c825b11987/Projects/STM32H7B3I-DK/Applications/FatFs/FatFs_uSD_DMA_Standalone/FatFs/Target/sd_diskio.c#L203
 Secondly there are the write and read Transfer completed callbacks.
https://github.com/STMicroelectronics/STM32CubeH7/blob/43c9e552ba1c038577c48723d96ca8c825b11987/Projects/STM32H7B3I-DK/Applications/FatFs/FatFs_uSD_DMA_Standalone/FatFs/Target/sd_diskio.c#L295

So, please allow me close this issue and thank you again for your contribution.

Best Regards,

gHuwk, 2024-04-24:
Hello. Thank you for your support of the product. I came across your closed Issue. To be honest, it is not entirely clear why the functions you specified use waiting for the operation to complete, which slightly contradicts the DMA mode. In fact, it's poling, only slightly improved, which is unclear in what sense it makes sense. I don't use FreeRTOS, using the main loop I have extremely bad write speeds - 30 MB of data arrives in 30 seconds, forcing the processor to wait until the callback is called. Are there any other options to solve the problem? I want to be able to execute code in other places while data is being transferred over SD.

https://github.com/STMicroelectronics/STM32CubeH7/blob/43c9e552ba1c038577c48723d96ca8c825b11987/Projects/STM32H7B3I-DK/Applications/FatFs/FatFs_uSD_DMA_Standalone/FatFs/Target/sd_diskio.c#L160

Perhaps I haven't read the forums completely, that's why I haven't come across a solution to the problem.
Thank you!

closed 2024-10-08 by ALABSTM