STMicroelectronics/STM32CubeH7 #246: Error comment in "system_stm32h7xx. c" file

chenjingyuanku, 2022-11-25:
The comment near the macro definition of "**VECT_TAB_BASE_ADDRESS**" in the "**system_stm32h7xx. c**" file says "**This value must be a multiple of 0x300.**" It should actually be **0x200**, which is a very misleading comment.

carlo-dev-git, 2022-12-10:
@ALABSTM I think this is duplicate of https://github.com/STMicroelectronics/cmsis_device_h7/issues/2
and it was fixed in release V1.11.0

TOUNSTM, 2023-02-23:
Hello,

Thank you for your report, but as @carlo-dev-git said  This issue has been previously reported by other users and it was fixed.
Please allow me to close this issue. Thank you again for your comprehension and for your contribution.

Best Regards

closed 2023-02-23 by TOUNSTM