STMicroelectronics/STM32CubeH7 #250: Whether ADC Vref channel and temperature acquisition channel are consistent with channel 18 and 19

wdfk-prog, 2022-12-20:
**Caution**
- Whether ADC Vref channel and temperature acquisition channel are consistent with channel 18 and 19
- In my actual tests, I found that using the Vref channel and temperature channel can output normal data
- However, using channels 18 and 19 does not generate the correct data

**Describe the set-up**
 * ART-PI STM32H750
 *KEIL5 AC6 -OZ

**Additional context**
- As clearly stated in the data sheet,18 channels are connected to temperature.19 channels are connected to internal reference voltage
- I used ADC Vref channel and temperature,It can be seen that the channel values are not the same.
```shell
VRef channel is 3484942336.
Temp channel is 3414425600.
18 channel is 1266941952.
19 channel is 1337458688.
VRef  is 24028.
Temp is 13102.
VRef voltage is 3327 mV.
46, and 52 are Temperature in Degree C calculated using two different formulas.
```
- Channels 18 and 19 are used
```shell
VRef  is 9558.
Temp is 7066.
VRef voltage is 8365 mV.
140, and 170 are Temperature in Degree C calculated using two different formulas.
```

### issue
- Is it a problem that channels 18 and 19 cannot be used to correctly collect the internal voltage and temperature, or is this the design?
**Screenshots**
If applicable, add screenshots to help explain your problem.

ARIOSTM, 2022-12-20:
Hi @wdfk-prog,

In order to read Temperature or Vrefint using ADC, you need to enable each channel in the ADC_CCR register.

ADC_CCR_TSEN to enable the VSENSE channel for the temperature sensor.
ADC_CCR_VREFEN to enable the Vrefint channel

When you call HAL_ADC_ConfigChannel with ADC_CHANNEL_TEMPSENSOR or ADC_CHANNEL_VREFINT the HAL driver automatically enable the corresponding channel.
The temperature sensor and vrefint are not enabled by the HAL driver when Channel 18/19 are configured.

In your test, did you make sure ADC_CCR_TSEN and ADC_CCR_VREFEN were enabled ? 

With regards,

wdfk-prog, 2022-12-21:
> Hi @ARIOSTM 
## I looked at the register values in KEIL5 and found some issues
### After the HAL_ADCEx_Calibration_Start function is enabled, all capabilities are set to one
```c
 if (HAL_ADCEx_Calibration_Start(&hadc3, ADC_CALIB_OFFSET_LINEARITY, ADC_SINGLE_ENDED) != HAL_OK) 
    { 
      Error_Handler();
    }
```
- Below is a screenshot of register values for ADC3 in KEIL5
[image]

- But I only configured the temperature and internal reference voltage channels
[image]

- The register value does not change when debug stops at the following location
[image]

### I don't know if that's a problem
- Why does configuring a channel not enable flag bits?
- Why is the flag bit enabled when the calibration function is enabled?

## The result of my attempt to comment out the calibration function is as follows

[image]

- None of the flags are in place, but I still have the reference voltage and temperature values I read
[image]

- The configuration code I use is as follows
```c
ADC_HandleTypeDef hadc3;
DMA_HandleTypeDef hdma_adc3;

/* ADC3 init function */
void MX_ADC3_Init(void)
{

  /* USER CODE BEGIN ADC3_Init 0 */

  /* USER CODE END ADC3_Init 0 */

  ADC_ChannelConfTypeDef sConfig = {0};

  /* USER CODE BEGIN ADC3_Init 1 */

  /* USER CODE END ADC3_Init 1 */

  /** Common config
  */
  hadc3.Instance = ADC3;
  hadc3.Init.Resolution = ADC_RESOLUTION_16B;
  hadc3.Init.ScanConvMode = ADC_SCAN_ENABLE;
  hadc3.Init.EOCSelection = ADC_EOC_SEQ_CONV;
  hadc3.Init.LowPowerAutoWait = DISABLE;
  hadc3.Init.ContinuousConvMode = ENABLE;
  hadc3.Init.NbrOfConversion = 2;
  hadc3.Init.DiscontinuousConvMode = DISABLE;
  hadc3.Init.ExternalTrigConv = ADC_SOFTWARE_START;
  hadc3.Init.ExternalTrigConvEdge = ADC_EXTERNALTRIGCONVEDGE_NONE;
  hadc3.Init.ConversionDataManagement = ADC_CONVERSIONDATA_DMA_CIRCULAR;
  hadc3.Init.Overrun = ADC_OVR_DATA_OVERWRITTEN;
  hadc3.Init.LeftBitShift = ADC_LEFTBITSHIFT_NONE;
  hadc3.Init.OversamplingMode = ENABLE;
  hadc3.Init.Oversampling.Ratio = 1024;
  hadc3.Init.Oversampling.RightBitShift = ADC_RIGHTBITSHIFT_10;
  hadc3.Init.Oversampling.TriggeredMode = ADC_TRIGGEREDMODE_SINGLE_TRIGGER;
  hadc3.Init.Oversampling.OversamplingStopReset = ADC_REGOVERSAMPLING_CONTINUED_MODE;
  if (HAL_ADC_Init(&hadc3) != HAL_OK)
  {
[... 142 lines cut ...]
    __HAL_ADC_CALC_TEMPERATURE(vref_mv, temp_value, ADC_RESOLUTION_16B));
    return RT_EOK;
}
```
## In addition, I manually modified the configuration code to use the 18/19 channel

[image]

[image]

- Calibration enabled, all flags at position one
[image]

- The results of uncalibration are as follows

[image]

[image]

## Finally, a graph is attached comparing the operating temperature and reference voltage to the register values using the 18/19 channel. You can see that the register values are the same
-  temperature and reference voltage 
[image]

- 18/19 channel
[image]

RJMSTM, 2023-08-08:
Hello @wdfk-prog 

Please allow me to close this thread as no activity. You may reopen it at any time if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards

closed 2023-09-05 by ALABSTM