STMicroelectronics/STM32CubeH7 #252: Potential buffer overflow in WiFi SPI driver

shijiameng, 2023-01-24:
**Bug Description**

- Tag: v1.11.0
- Buggy code: https://github.com/STMicroelectronics/STM32CubeH7/blob/v1.11.0/Projects/STM32H7B3I-DK/Demonstrations/ClockAndWeather/Core/Src/net_conf_es_wifi_spi.c#L296

```C
static int8_t SPI_WIFI_ResetModule(void)
{
    uint32_t tickstart = HAL_GetTick();
    uint8_t Prompt[6];
    uint8_t count = 0;
    HAL_StatusTypeDef  Status;

    WIFI_RESET_MODULE();
    WIFI_ENABLE_NSS();
    SPI_WIFI_DelayUs(10);

    while (WIFI_IS_CMDDATA_READY())
    {
        Status = HAL_SPI_Receive(&hspi, &Prompt[count], 1, 0xFFFF);  // !!! Buggy line: bound check is missing !!!
        count += 2;
        if (((HAL_GetTick() - tickstart) > 0xFFFF) || (Status != HAL_OK))
        {
            WIFI_DISABLE_NSS();
            return -1;
        }
    }

    WIFI_DISABLE_NSS();
    if ((Prompt[0] != 0x15) || (Prompt[1] != 0x15) || (Prompt[2] != '\r') ||
            (Prompt[3] != '\n') || (Prompt[4] != '>') || (Prompt[5] != ' '))
    {
        return -1;
    }
    return 0;
}
```

- Description: In function ``SPI_WIFI_ResetModule``, it invokes ``HAL_SPI_Receive()`` to receive data from SPI interface and places the received data to array ``prompt`` at the index specified by ``count``, then it adds ``count`` with 2.  Bound check on array ``prompt`` is missing, which may cause stack buffer overflow and corrupt ``tickstart`` if ``count`` exceeds the length of ``prompt``.

**Patch Suggestion**

Add bound check before invoking ``HAL_SPI_Receive()``.

```C
static int8_t SPI_WIFI_ResetModule(void)
{
    uint32_t tickstart = HAL_GetTick();
    uint8_t Prompt[6];
    uint8_t count = 0;
    HAL_StatusTypeDef  Status;

    WIFI_RESET_MODULE();
    WIFI_ENABLE_NSS();
    SPI_WIFI_DelayUs(10);

    while (WIFI_IS_CMDDATA_READY())
    {
        // Patch: bound check before receiving data from SPI
        if (count == 4) {
            WIFI_DISABLE_NSS();
            return -1;
        }

        Status = HAL_SPI_Receive(&hspi, &Prompt[count], 1, 0xFFFF);
        count += 2;
        ...
    }
    ...
    return 0;
}
```

ALABSTM, 2023-08-01:
ST Internal Reference: 158546

ALABSTM, 2023-08-01:
Hi @shijiameng,

Thank you for this report too.

With regards,