STMicroelectronics/STM32CubeH7 #254: New STM32H747-DISCO - BSP_LCD_Init() Issue?

AndrewCapon, 2023-03-08:
Hi Guys,

I have a new STM32H747-DISCO board here, I have been to the ST Community and there is another guy with a new board that is seeing the same issue: https://community.st.com/s/question/0D53W00002A3uJaSAJ/stm32h747-lcd-display-issue

So for most of the examples using the LCD I see this:

[image]

If I load up one of the CmdMode examples for instance `LCD_DSI_CmdMode_SingleBuffer` then the screen is working fine, it seems to be handling the DSI transfers itself.

If I load up any example that is using `BST_LCD_Init()" for instance `LCD_DSI_VideoMode_SingleBuffer` then the screen just show the unmoving static as shown in the image attached.

I have stepped through the code of `LCD_DSI_VideoMode_SingleBuffer` checking that there are no error returns anywhere and everything looks ok, DMA is working, data is in the frame buffer but it just seems the frame buffer is not being pushed to the screen.

I'm wondering if maybe there is a new revision of this board and the existing code is not quite working.

Has anyone any ideas of what to check for here?

Many thanks

Andy

AndrewCapon, 2023-03-21:
Hi @HBOSTM,

Super thanks for that.

Andy

AndrewCapon, 2023-05-02:
Hi @HBOSTM 

Do you have an update on this?

More and more people are popping up on the forum with issues with these displays.

Andy

AndrewCapon, 2023-06-05:
Hi @HBOSTM 

Thanks for the info, I had already been informed on the forum and have done a quick test. The examples included are incorrect as they do not include the new components, when that is fixed they now work.

There may still be some confusion from users though as the boards that are not working before have an A3 sticker not an A9.

s1d-r, 2024-02-20:
Hi @HBOSTM and @AndrewCapon  I have changed the drivers and everything yet its not working for me
this is set of errors and warnings I am receiving 

arm-none-eabi-gcc "/home/ad/en.FP-AI-VISION1_V3.1.0/FP-AI-VISION1_V3.1.0/Drivers/BSP/STM32H747I-Discovery/stm32h747i_discovery_lcd.c" -mcpu=cortex-m7 -std=gnu11 -g3 -DUSE_HAL_DRIVER -DSTM32H747xx -DUSE_STM32H747I_DISCO -DCORE_CM7 -DDATA_IN_ExtSDRAM -DMEMORY_SCHEME=3 -DCAMERA_CAPTURE_RES=2 -DPIXEL_FMT_CONV=1 -DARM_MATH_CM7 -DOBJECTDETECT -DSTM32IPL -D__FPU_PRESENT=1 -c -I../../../CM7/Inc -I../../../../Common/CM7/Inc -I../../../../../../../Drivers/CMSIS/Device/ST/STM32H7xx/Include -I../../../../../../../Drivers/CMSIS/Include -I../../../../../../../Drivers/CMSIS/DSP/Include -I../../../../../../../Drivers/STM32H7xx_HAL_Driver/Inc -I../../../../../../../Drivers/BSP/STM32H747I-Discovery -I../../../../../../../Drivers/BSP/Components/Common -I../../../../../../../Drivers/BSP/Components/ov9655 -I../../../../../../../Drivers/BSP/Components/otm8009a -I../../../../../../../Utilities/Fonts -I../../../../../../../Utilities/lcd -I../../../../../../../Middlewares/ST/STM32_AI_Runtime/Inc -I../../../../../../../Middlewares/ST/STM32_AI_Utilities/Inc -I../../../../../../../Middlewares/ST/STM32_ImageProcessing_Library/Inc -I../../../../../../../Middlewares/Third_Party/FatFs/src -I../../../CM7/lib/libpeopledetect/Inc -Ofast -ffunction-sections -fdata-sections -Wall -fstack-usage -fcyclomatic-complexity -MMD -MP -MF"Drivers/BSP/STM32H747I_DISCO/stm32h747i_discovery_lcd.d" -MT"Drivers/BSP/STM32H747I_DISCO/stm32h747i_discovery_lcd.o" --specs=nano.specs -mfpu=fpv5-d16 -mfloat-abi=hard -mthumb -o "Drivers/BSP/STM32H747I_DISCO/stm32h747i_discovery_lcd.o"
arm-none-eabi-gcc -o "STM32H747I-DISCO_PeopleCounting_CM7.elf" @"objects.list"  -l:LibPeopleDetect110_CM7_GCC.a -l:NetworkRuntime710_CM7_GCC.a -mcpu=cortex-m7 -T"/home/ad/en.FP-AI-VISION1_V3.1.0/FP-AI-VISION1_V3.1.0/Projects/STM32H747I-DISCO/Applications/PeopleCounting/STM32CubeIDE/CM7/STM32H747XIHx_FLASH_CM7_Full_Internal_Fps.ld" --specs=nosys.specs -Wl,-Map="STM32H747I-DISCO_PeopleCounting_CM7.map" -Wl,--gc-sections -static -L../../../../../../../Middlewares/ST/STM32_AI_Runtime/lib -L../../../CM7/lib/libpeopledetect -u _printf_float --specs=nano.specs -mfpu=fpv5-d16 -mfloat-abi=hard -mthumb -Wl,--start-group -lc -lm -Wl,--end-group
/opt/st/stm32cubeide_1.14.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.linux64_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /opt/st/stm32cubeide_1.14.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.linux64_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+dp/hard/libc_nano.a(libc_a-closer.o): in function `_close_r':
(.text._close_r+0xc): warning: _close is not implemented and will always fail
/opt/st/stm32cubeide_1.14.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.linux64_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /opt/st/stm32cubeide_1.14.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.linux64_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+dp/hard/libc_nano.a(libc_a-fstatr.o): in function `_fstat_r':
(.text._fstat_r+0xe): warning: _fstat is not implemented and will always fail
/opt/st/stm32cubeide_1.14.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.linux64_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /opt/st/stm32cubeide_1.14.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.linux64_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+dp/hard/libc_nano.a(libc_a-signalr.o): in function `_getpid_r':
(.text._getpid_r+0x0): warning: _getpid is not implemented and will always fail
/opt/st/stm32cubeide_1.14.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.linux64_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /opt/st/stm32cubeide_1.14.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.linux64_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+dp/hard/libc_nano.a(libc_a-isattyr.o): in function `_isatty_r':
(.text._isatty_r+0xc): warning: _isatty is not implemented and will always fail
/opt/st/stm32cubeide_1.14.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.linux64_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /opt/st/stm32cubeide_1.14.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.linux64_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/lib/thumb/v7e-m+dp/hard/libc_nano.a(libc_a-signalr.o): in function `_kill_r':
(.text._kill_r+0xe): warning: _kill is not implemented and will always fail
/opt/st/stm32cubeide_1.14.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3.rel1.linux64_1.1.100.202309141235/tools/bin/../lib/gcc/arm-none-eabi/11.3.1/../../../../arm-none-eabi/bin/ld: /opt/st/stm32cubeide_1.14.0/plugins/com.st.stm32cube.ide.mcu.externaltools.gnu-tools-for-stm32.11.3 [... cut ...] 32H747I_DISCO/stm32h747i_discovery_lcd.o: in function `BSP_LCD_InitEx':
/home/ad/en.FP-AI-VISION1_V3.1.0/FP-AI-VISION1_V3.1.0/Drivers/BSP/STM32H747I-Discovery/stm32h747i_discovery_lcd.c:404: undefined reference to `NT35510_LCD_Driver'
collect2: error: ld returned 1 exit status
make: *** [makefile:71: STM32H747I-DISCO_PeopleCounting_CM7.elf] Error 1
"make -j6 all" terminated with exit code 2. Build might be incomplete.

10:46:34 Build Failed. 4 errors, 8 warnings. (took 1s.442ms)

[image]
[image]

ALABSTM, 2024-03-14:
Hi everyone,

Please excuse this delayed reply.
* @AndrewCapon
  *  To answer your question about the `A03` and `A09` labels, it is `A09` indeed. Board revision `A03` is no more active and has been replaced by the newer revision `A09`. However, the same recommendations apply to both.
  * You mentioned some examples that do not include the new `NT35510` component's driver yet. You are right. We identified this list. This will be fixed soon hopefully.
    * `./Projects/STM32H747I-DISCO/Applications/FPU/FPU_Fractal`
    * `./Projects/STM32H747I-DISCO/Examples/DCMI/DCMI_SnapshotMode`
    * `./Projects/STM32H747I-DISCO/Examples/DMA/DMAMUX_RequestGen`
    * `./Projects/STM32H747I-DISCO/Examples/MDMA/MDMA_DMA2D_Triggering`
    * `./Projects/STM32H747I-DISCO/Examples/MDMA/MDMA_LinkedList_ColorsComp`
    * `./Projects/STM32H747I-DISCO/Examples/MDMA/MDMA_RepeatBlock_Rotation`
    * `./Projects/STM32H747I-DISCO/Examples/MDMA/MDMA_RepeatBlock_ZoomOut`

* @dc5ala, a new way to dynamically identify the LCD screen has been introduced since commit dd1b1d7144e6c61f995a368bbbeaad1936d60cd1. Hence, there is no more need for you to update the values of macros `USE_LCD_CTRL_OTM8009A` and `USE_LCD_CTRL_NT35510`. The comment below is no more relevant and shall be removed in a future version of this firmware.

https://github.com/STMicroelectronics/STM32CubeH7/blob/dd1b1d7144e6c61f995a368bbbeaad1936d60cd1/Drivers/BSP/STM32H747I-DISCO/stm32h747i_discovery.h#L22-L25

* @s1d-r, it looks from the screenshots you shared that you have not included the `NT35510` driver's source files in your project's configuration file.

With regards,

ALABSTM, 2024-03-14:
ST Internal Reference: 175742

ALABSTM, 2024-03-14:
ST Internal Reference: 176107

ALABSTM, 2024-03-27:
Hi @AndrewCapon and @dc5ala,

The remaining issues reported have been fixed in the frame of release **1.11.2**. Please allow me to close this thread.

With regards,

closed 2024-03-27 by ALABSTM

AndrewCapon, 2024-03-27:
Thanks @ALABSTM

ALABSTM, 2024-05-27:
You're welcome @AndrewCapon.