STMicroelectronics/STM32CubeH7 #256: H730 ADC3 Oversampling configuration bug

KubeckPL, 2023-04-19:
**Caution**
There is an error in the oversampling configuration for the ADC3 in the H730VBT processor in the HAL library.
In the attached screenshots, I have shown the configuration in CubeMX, in the ADC3 handler, and the final values written to the ADC3 registers.
Tools used:
- STM32CubeIDE V1.12,
- HAL Driver version number V1.11.1

**Screenshots**

CubeMX ADC oversampling configuration:
[image]

MX_ADC3_Init() function:
[image]

Final value in CFGR2 register (equal to 4):
[image]

**Notice**
Oversampling ratios for ADC3 in the H730 MCU are predefined, so CubeMX should only propose predefined values:

[image]

KubeckPL, 2023-04-19:
The proposed solution in the MX_ADC3_Init() function - using a predefined macro:

[image]

RJMSTM, 2023-04-20:
Hello @KubeckPL,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension. 

With regards,
Rania

RJMSTM, 2023-08-07:
Hello @KubeckPL,

Could you please share the code you have used to reproduce the problem?

Regards

RJMSTM, 2024-01-15:
Hello @KubeckPL, 
Any update please, you resolve your problem??

KubeckPL, 2024-01-15:
Hi @RJMSTM.
The bug is still present in the latest version of CubeIDE and CubeMX (V1.14) - I have just checked this.
This problem relates to the code generator built into CubeMX.

[image]

RJMSTM, 2024-01-15:
Hello @KubeckPL,

Thank you for this report. The point you reported is actually related to the CubeMX generation problem and not to the firmware published in this repository. Unfortunately, we don't treat aspect related to CubeMX tool in our GitHub repositories. They are rather treated at the [STM32CubeMX](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) dedicated page of the ST Community .

Since this issue is not directly related to the STM32Cube firmware but rather to our ecosystem, please allow me then to close this thread. Thank you for your comprehension.

With regards,

closed 2024-05-27 by ALABSTM