STMicroelectronics/STM32CubeH7 #258: Octo-SPI multiplexed mode unavailable

dennisvbussel, 2023-05-02:
**Describe the set-up**
 * STM32H730VBTx
 * STM32CubeIDE 6.8.1

**Describe the bug**
OCTOSPI multiplexed mode cannot be selected in STM32CubeMX. According to the reference manual the OCTOSPI in multiplexed mode should be available in the STM32H730 Value line (RM0468). Also AN5050 lists STM32H730 value line in the Applicable products table.

See image below:
[image]

Could this be a bug in the configurator, or am I missing something?

**Additional context**
I found an example project in STM32CubeMX (STM32L4P5G-DK) where the OSPI multiplexed mode can be selected.

HBOSTM, 2023-05-16:
Hello @dennisvbussel ,

The issue you reported has been logged into an internal bug tracker, it will be managed by the development team.

However, as it is related to the ecosystem (namely the CubeMX tool) rather than the firmware published within this repository (CMSIS, HAL, BSP, etc.), please allow me thus to keep this thread closed.

In case you have other CubeMX-related issues to share, please submit them in the [ST Community](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) in the CubeMX related section. There you shall find people who will address your request and whom you will hopefully find a solution with.

Since this issue is not directly related to the STM32Cube firmware, please allow me then to close this thread. Thank you for your comprehension.

With regards,

HBOSTM, 2023-05-16:
ST Internal Reference: 152182

closed 2023-05-16 by HBOSTM