STMicroelectronics/STM32CubeH7 #261: Ethernet low_level_init does not handle return code from LAN8742_init

JojoS62, 2023-06-26:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus) 

**Describe the set-up**
 * NUCLEO_H743ZI2 Rev. E01
 * STM32Cube FW_H7 V1.11.0

**Describe the bug**
The return code from https://github.com/STMicroelectronics/STM32CubeH7/blob/c2b13142a354d26b0baa5f505bb5607cd4aadd88/Projects/NUCLEO-H743ZI/Applications/LwIP/LwIP_HTTP_Server_Netconn_RTOS/Src/ethernetif.c#L216C10-L216C10

is not used. I have the case that a timeout error occurs. The code continues without checking (same in CubeMX generated code) and the link status is set wrongly to 10 M / half duplex. A following dhcp sequence executes correctly, but never sends a valid discover on the wire. So the debugging is difficult, as everything seems to be fine.
How can the soft reset of LAN8742 fail? Haven't tested yet, but must not the link mode set before the soft reset? According to the datasheet, the soft reset doesn't evaluate the hardware strapping bits.

**How To Reproduce**
1. Indicate the global behavior of your application project
Build a ethernet app with CubeMX (needs additional steps, CubeMX code is not valid at the moment) or use the LwIP_HTTP_Server_Netconn example.

3. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...)
Middlewar LWIP, ethernetif.c

5. The use case that generates the problem
power on board with external supply

6. How we can reproduce the problem
I can reproduce the problem by using an external 5 V power supply, or unplug/plug the JP2 in STLink position

**Additional context**
return code of LAN8742 must be checked. In a debug build, it should run into ErrorHandler. 
Print out the Error Codes as in lwip, there is a lot of nice tracing output. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

JojoS62, 2023-07-06:
thanks for your response.
I had also opened a thread in the ST community:
https://community.st.com/t5/stm32cubemx-mcu/h743-and-lan8742-init/td-p/570179

Two more issues with the LAN8742_init:
- I found that the H7 is very fast, the init is called about 3 ms after power up. But the chip needs more time to initialize and to read the strapping bits. A delay of 50 ms before calling the lwip init fixed the problem. For cube generated code, the MX_LWIP_init has to be disabled from automatic calling for this.
- the LAN8742 init tries addresses 0..31, but with its strapping bits it can have only address 0 or 1. It is more important to verify if the SMR read was successful, then the delay is not necessary.

TOUNSTM, 2023-08-15:
ST Internal Reference: 159319

ALABSTM, 2024-02-21:
Fixed in commit ae3bb5433e3a493b8d759ccd39dc50dce4e49e30

closed 2024-02-21 by ALABSTM