STMicroelectronics/STM32CubeH7 #262: USB_Host driver: potential buffer overflow when parsing configuration descriptor

shijiameng, 2023-06-26:
**Describe the set-up**
 * Board: stm32h743i-eval
 
**Describe the bug**

In source file [usblctreq.c](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_ctlreq.c), the function `USBH_ParseCfgDesc` is responsible for parsing the configuration description of USB device. At [line 507](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_ctlreq.c#L507), the number of endpoint descriptors to be parsed depends on `bNumEndpoints` field as shown in [line 485](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Middlewares/ST/STM32_USB_Host_Library/Core/Src/usbh_ctlreq.c#L485) and each endpoint is indexed by `ep_ix`. However, the driver assumes that the maximum number of endpoints per interface is 2 (defined by macro `USBH_MAX_NUM_ENDPOINTS`). Therefore, a buffer overflow on array `Ep_Desc` would arise at line 507 if a malicious USB device responds more than `USBH_MAX_NUM_ENDPOINTS` (2) endpoint descriptors (i.e., `bNumEndpoints` >= `USBH_MAX_NUM_ENDPOINTS`).

shijiameng, 2023-07-02:
I found this issue has been revised (CVE-2021-42553). Close it

closed 2023-07-02 by shijiameng

ALABSTM, 2023-07-12:
Hi @shijiameng,

Thank you for your report. The issue you pointed out has been fixed in the frame of version **3.5.1** of our USB Host library, available [here](https://github.com/STMicroelectronics/stm32_mw_usb_host/tree/38bc365e09b89c43b495af7261e31fe90a07335c).

Currently, the STM32CubeH7 firmware still integrates version **3.5.0** of this library. However, you can easily download the newer version and integrate it manually into your package.

With regards,