STMicroelectronics/STM32CubeH7 #267: USBH_MSC_BOT_Process: Too much byte are sent with USB_EP_DIR_OUT

jbiemar, 2023-07-13:
**Describe the set-up**
 * NUCLEO-H723ZG NUH723ZG$AT2
 * STM32CubeIDE v1.12.1

**Describe the bug**
In the function `USBH_MSC_BOT_Process` from file `Middlewares\ST\STM32_USB_Host_Library\Class\MSC\Src\usbh_msc_bot.c` the length of sent data may be too long in some case.
```
    case BOT_DATA_OUT:
      (void)USBH_BulkSendData(phost, MSC_Handle->hbot.pbuf,
    		  	  	  MSC_Handle->OutEpSize, MSC_Handle->OutPipe, 1U);

      MSC_Handle->hbot.state  = BOT_DATA_OUT_WAIT;
      break;
```
With a custom SCSI device, the command is misunderstood and the device never respond because we are sending 64 bytes instead of 36 expected in the variable `MSC_Handle->hbot.cbw.field.DataTransferLength`.

**How To Reproduce**
Use the `USBH_StatusTypeDef USBH_MSC_BOT_Process(USBH_HandleTypeDef *phost, uint8_t lun)` function.

**Additional context**
My way to fix this is to modify function to this:
```
    case BOT_DATA_OUT:
      (void)USBH_BulkSendData(phost, MSC_Handle->hbot.pbuf,
    		  	  	  (MSC_Handle->hbot.cbw.field.DataTransferLength > MSC_Handle->OutEpSize) ? MSC_Handle->OutEpSize : MSC_Handle->hbot.cbw.field.DataTransferLength,
    				  MSC_Handle->OutPipe, 1U);

      MSC_Handle->hbot.state  = BOT_DATA_OUT_WAIT;
      break;
```

RJMSTM, 2023-07-21:
Hello @jbiemar,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,

ALABSTM, 2024-03-08:
ST Internal Reference: 175667

ALABSTM, 2024-03-11:
Hi @jbiemar,

Back to you about this point. Our development teams acknowledged your proposal and its relevance. However, they reported there is **no plan** to support such enhancement for the moment.

Please allow me to close this thread. Thank you for your comprehension.

With regards,

closed 2024-03-11 by ALABSTM