STMicroelectronics/STM32CubeH7 #268: VECT_TAB_OFFSET comments in different releases

escherstair, 2023-07-25:
In file system_stm32h7xx.c the macro VECT_TAB_OFFSET is defined in different ways under several conditions.
This is ok.
But I see a comments on these definition, that changes from release to release, and I don't understand why.
You can see it blaming the file [system_stm32h7xx.c](https://github.com/STMicroelectronics/STM32CubeH7/blame/master/Drivers/CMSIS/Device/ST/STM32H7xx/Source/Templates/system_stm32h7xx.c).
In release 1.9.1 it was
```
#define VECT_TAB_OFFSET         0x00000000U       /*!< Vector Table base offset field.
                                                       This value must be a multiple of 0x200. */
```
note the value `0x200`
In release 1.10.0 it was
```
#define VECT_TAB_OFFSET         0x00000000U       /*!< Vector Table base offset field.
                                                       This value must be a multiple of 0x300. */
```
note that now the value is `0x300`
In release 1.11.0 it is
```
#define VECT_TAB_OFFSET         0x00000000U       /*!< Vector Table base offset field.
                                                       This value must be a multiple of 0x400. */
```
note the value `0x400`

I dont' understand the reason for this change.
Similar changes in the comment can be seen for `VECT_TAB_BASE_ADDRESS`

I need to relocate the vector table, and so I must be sure which is the allowed values.
And most importan idf they change with releases on the firmware.
I have an application developed with 1.9.1 and my offset is a multiple of `0x200` (but not a multiple of `0x400`)
Should I change it to a different value (multiple of `0x400`) if I migrate my project to firmware 1.11.0?

TOUNSTM, 2023-08-02:
Hello @escherstair,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With Regards,

carlo-dev-git, 2023-08-09:
Please check this too https://github.com/STMicroelectronics/cmsis_device_h7/issues/2

ALABSTM, 2023-08-11:
Hi,

@escherstair, the offset multiple to consider is `0x400` indeed.  As suggested by @carlo-dev-git, the confirmation can be found in issue [cmsis_device_h7#2](https://github.com/STMicroelectronics/cmsis_device_h7/issues/2#issuecomment-1094765954).

The rationale is that the vector table offset shall:
* be a multiple of its total size in bytes **rounded to the nearest** power of 2,
* comply to the constraint that **bits 0 to 8 of the VTOR** register are reserved, thus shall be **left to zero**.

As for STM32H7, we have `162 lines + 16 exceptions = (162 + 16) x 4 = 0x2C8` --> rounded to `0x400`.

You may find further details in the [programmers manual](https://www.st.com/resource/en/programming_manual/pm0253-stm32f7-series-and-stm32h7-series-cortexm7-processor-programming-manual-stmicroelectronics.pdf), revision 5, page 197, paragraph 4.3.4.

I hope this answers your question. Please allow me to close this thread.

**PS:** @carlo-dev-git, thank you very much for you helpful reply.

closed 2023-08-11 by ALABSTM