STMicroelectronics/STM32CubeH7 #271: is42s32800j_conf_template.h include "stm32f7xx_hal" path error.

KwangryeolPark, 2023-08-21:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus) 

**Describe the set-up**
 * STM32H747I-DISCO
 * STM32CubeIDE 1.12.1 version and STM32Cube_FW_H7_V1.11.1

**Describe the bug**
```fatal error: stm32f7xx_hal.h: No such file or directory``` when including stm32h747i_discovery_sdram.h at main.c

**How To Reproduce**
1. Indicate the global behavior of your application project
1.1. include stm32h747i_discovery_sdram.h at main.c
1.2. modify the conf_template.h files to conf.h

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...)
At "BSP/Components/is42s32800j/is42s32800j_conf_template.h", the include path contains **stm32f7xx_hal.h** not **stm32h7xx_hal.h**

TOUNSTM, 2023-10-02:
ST Internal Reference: 162782

TOUNSTM, 2023-10-05:
Hello @KwangryeolPark,

Unfortunately, our development team has declined your proposal. According to them, it's up to the user to copy/paste the `is42s32800j_conf_template.h` file into his application and customize it as he sees fit. Even with "stm32xxxx_hal.h" include the customer will get a compile error and it's up to him to correct it.
Please allow me to close this thread. Thank you for your understanding.

With Regards,

closed 2023-10-05 by TOUNSTM