STMicroelectronics/STM32CubeH7 #272: HAL_UART_Transmit can generate faulty error

ra1u, 2023-08-29:
Function [`HAL_UART_Transmit()`](https://github.com/STMicroelectronics/STM32CubeH7/blob/91d4c89228e766a968f8daebc78eb7143d95184e/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_uart.c#L1120) can return HAL_TIMEOUT error that should not be returned.

Issue is that HAL_UART_Transmit() uses [`UART_WaitOnFlagUntilTimeout()`](https://github.com/STMicroelectronics/STM32CubeH7/blob/91d4c89228e766a968f8daebc78eb7143d95184e/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_uart.c#L3414) 

In case that there is rx timeout (data was regularly received during transmission and rx timout isr is raised) [HAL_TIMEOUT](https://github.com/STMicroelectronics/STM32CubeH7/blob/91d4c89228e766a968f8daebc78eb7143d95184e/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_uart.c#L3463) is returned during transmission.

TOUNSTM, 2023-08-29:
Hello @ra1u,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With Regards,

ASELSTM, 2023-09-06:
Hi @ra1u,

Would it be possible to share the whole project you have used to reproduce the issue in order to allow a better analysis of the problem.

With regards,

ra1u, 2023-09-06:
I am sorry I cant share project,  but out of my head it should be following.

// config
Config uart 
set rx timeouts ->` HAL_UART_ReceiverTimeout_Config(UART_HandleTypeDef *huart, uint32_t TimeoutValue)`
enable rx timeout -> `HAL_StatusTypeDef HAL_UART_EnableReceiverTimeout(UART_HandleTypeDef *huart)`

In our case we use customised uart ISR handler, however issue remains.

We would expect, that timeouts for RX and TX are to be handled separately and that Rx timeout does not trigger error in `UART_WaitOnFlagUntilTimeout()` when waiting for TX to complete.
Our internal patch for this is that we added additional argument for `UART_WaitOnFlagUntilTimeout `that distinguish direction. (it is equivalent as replacing `UART_WaitOnFlagUntilTimeout  `with two functions `UART_WaitOnFlagUntilTimeout`_RX and `UART_WaitOnFlagUntilTimeout`_TX )

KRASTM, 2024-02-26:
Hello @ra1u,

I want to clarify the issue; in your project you used HAL_UART_Transmit() to transmit Data and for the receiving part you used an interruption to receive data HAL_UART_Receive_IT()?
if not, can you share more information about your config & project, so we can reproduce it.

KRASTM, 2024-02-26:
Hello @ra1u,

Just to be sure, in your project you used HAL_UART_Transmit() to transmit Data, and for the receiving part you used an ISR to receive data for ex: HAL_UART_Receive_IT()?

Can you share more information about your config & project, so we can reproduce it.

With regards

KRASTM, 2024-03-05:
Hello @ra1u,

Any news please?

With regards.

KRASTM, 2024-03-25:
Hello @ra1u,

As a result of the STM32CubeH7 being recently updated to v1.11.2, can you please verify that the issue still remains.

With regards.

KRASTM, 2024-04-04:
Hello @ra1u,

Please allow me to close the issue, since there is no activity in a while.

With regards.

closed 2024-04-04 by KRASTM