STMicroelectronics/STM32CubeH7 #274: USB composite CDC-CustomHID device problem, one of I/F hangs or data corruption occurs

danielsulik, 2023-09-07:
**Brief description**
A serious problem with USB composite CDC-CustomHID device working reliability. I have ported your H7 Composite Custom HID - CDC example code to STM32F723 MCU. 
HID interface either hangs, sometimes CDC I/F hangs or data integrity fails when running loopback test on both 
interfaces at the same time from two different host process.
 
**Describe the set-up**

The board: Not applicable. I was testing with STM32F723-DISCO development board.
IDE: System Workbench for STM32
Observed on Windows 7, 8.1, 10

**Describe the bug**
I have implemented USB composite CDC- and Custom HID device based on your H7 Composite Custom HID - CDC example code.
HID I/F receives 64 bytes long reports and sends the same reports back to the host running a HID loopback test process .
CDC I/F receives random bytes long packets from length between 1 to 12345 bytes) and sends the same data back to the host from a different test process running on the same host.
Device enumerates OK with PID 0x483 an PID 0x5741, 
If I run test on one interface only it works reliably 247 without any problem.
Once I run loopback test on both interfaces one of the interfaces hangs or incorrect data is received by the host.
Does not matter if the code is running in HS or FS mode.
For simplicity no UART is implemented on CDC side. There is just a simple loopback. 

The main loop is not stuck, LED is still blinking, however processing of comms on USB either CDC or HID is stuck. Looks to me like missing an interrupt processing. 
Minimal version of the code is in the attachment.
[cdc-hid-min.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/12553923/cdc-hid-min.zip)

danielsulik, 2023-09-24:
To replicate the problem quicker set bInterval parameter to 0x1 on CDC and HID interface.
in usbd_customhid.h
#define CUSTOM_HID_HS_BINTERVAL                      0x1U
#define CUSTOM_HID_FS_BINTERVAL                      0x1U
in usbd_cdc.h
#define CDC_HS_BINTERVAL                            0x1U
#define CDC_FS_BINTERVAL                            0x1U

danielsulik, 2023-10-19:
HI there,
please is anyone looking at the issue. We need to sort this out asap.

ALABSTM, 2024-02-14:
Hi @danielsulik,

Our apology for this delayed reply. Your report has been forwarded to our development teams for investigation. I will keep you informed.

With regards,

ALABSTM, 2024-06-24:
Hi @danielsulik,

Are you still facing the same issue? Your report has been forwarded to our development teams. Let's wait for their feedback.

With regards,

ALABSTM, 2024-06-24:
ST Internal Reference: 184861

danielsulik, 2024-06-24:
Yes, still the same problem.

Regards

Daniel

ALABSTM, 2026-08-31:
Hi @danielsulik,

Thank you again for this report and for the inputs provided to help us reproduce the issue. Unfortunately, we were not able to, due to two files, `cdctest.tcl` and `hidtest.tcl`, that are missing in the provided zipped project, files that are mentioned in the `readme.txt` file.

Please allow me to close this issue. You can resubmit a new one, should the problem still persist. Thank you for your understanding.

With regards,

closed 2026-08-31 by ALABSTM