STMicroelectronics/STM32CubeH7 #275: STM32H7B3I-DK PLL3 VCO output exceeds 560MHz limit

sirius506, 2023-09-10:
STM32H7B3I-BK BSP LCD driver sets PLL3 VCO output to 800MHz.
I know it works, but Data sheet says maximum value is 560MHz.

https://github.com/STMicroelectronics/STM32CubeH7/blob/91d4c89228e766a968f8daebc78eb7143d95184e/Drivers/BSP/STM32H7B3I-DK/stm32h7b3i_discovery_lcd.c#L437-L450

TOUNSTM, 2024-03-07:
Hello @sirius506,

Sorry for the delayed response. Could you please send me the document in which you find this information?

With regards,

sirius506, 2024-03-08:
Please refer STM32H7B3I datasheet Table 55 on page 112. It says PLL VCO output max freq is 560MHz.

You can also try to set above listed PLL3 settings on the CubeMX clock config screen, then you'll see it responds with error indication (DIVN3 = 200 is too big).

TOUNSTM, 2024-03-27:
ST Internal Reference: 177140