STMicroelectronics/STM32CubeH7 #278: H7B3: USB with DMA generates many interrupts

PiPointX, 2023-09-19:
**Describe the set-up**
I'm using a H7B3 on a custom board. On this board I want to use USB with Azure USBx. 
The USB configuration in CubeMX is:

- USB host only
- DMA enabled
- internal PHY 
- Speed: FullSpeed

I'm using CubeIDE 1.10.1.
The USB device I want to use with the USB Host is a Prolific-Device.

**Describe the bug**
Regarding the Reference Guide, USB interrupts like NAK, NYET, ... should be handled automatically by the DMA. But when I use the confguration above the interrupts never get disabled in USB_HC_StartXfer. And the interrupts get fired very often and have a huge influence on the performance of the controller.

**How To Reproduce**
1. Set a breakpoint into the function USB_HC_StartXfer.
2. Connect a prolific device
3. Step through function

````
USBx_HC((uint32_t)ch_num)->HCINTMSK &= ~(USB_OTG_HCINTMSK_NYET |
                                               USB_OTG_HCINTMSK_ACKM |
                                               USB_OTG_HCINTMSK_NAKM);
````
4. The code block above gets neve reached because of this check here:

``
if (((USBx->CID & (0x1U << 8)) != 0U) && (hc->speed == USBH_HS_SPEED))
``

**Additional context**
It has to do with this issue here: https://community.st.com/t5/stm32-mcus-products/stm32f4-stm32f7-usb-host-core-interrupt-flood/td-p/436225

But the interrupts still get triggered even when DMA is enabled

**Screenshots**
If applicable, add screenshots to help explain your problem.

ALABSTM, 2025-07-31:
Hello @PiPointX,

Sorry for the delayed reply. You are right about this point. It has been **fixed** and the condition you mentioned has been removed as you can see [here](https://github.com/STMicroelectronics/stm32h7xx-hal-driver/commit/6dd7f0e35c80b6e46ab7b77c972d2e785054d7c6#diff-8ff0344e1def32f0cdab932c2006f8b69a8c337438ab47273aa1a58b340c2c0cL1792).

With regards,

ALABSTM, 2025-07-31:
Fixed in commit  37cbebdbc41f2fbbab3c57af0190ea097fe9b842

closed 2025-07-31 by ALABSTM