STMicroelectronics/STM32CubeH7 #283: STM32H725 and FreeRTOS doesn't generate timer IRQ correctly.

Makin-Things, 2024-01-09:
**Describe the set-up**
Using STM32CubeIDE 1.14.0 and FW)H& V1.11.1

**Describe the bug**
When you generate a project for the STM32H725 using FreeRTOS and setting the Timbase Source to TIM6 it generates a completely empty ISR for TIM6.

```
/**
  * @brief This function handles TIM6 global interrupt, DAC1_CH1 and DAC1_CH2 underrun error interrupts.
  */
void TIM6_DAC_IRQHandler(void)
{
  /* USER CODE BEGIN TIM6_DAC_IRQn 0 */

  /* USER CODE END TIM6_DAC_IRQn 0 */
  /* USER CODE BEGIN TIM6_DAC_IRQn 1 */

  /* USER CODE END TIM6_DAC_IRQn 1 */
}
```

When you do the same for a H723 it generates the correct code as below, which is what should also be generated for the H725.

```
/**
  * @brief This function handles TIM6 global interrupt, DAC1_CH1 and DAC1_CH2 underrun error interrupts.
  */
void TIM6_DAC_IRQHandler(void)
{
  /* USER CODE BEGIN TIM6_DAC_IRQn 0 */

  /* USER CODE END TIM6_DAC_IRQn 0 */
  HAL_TIM_IRQHandler(&htim6);
  /* USER CODE BEGIN TIM6_DAC_IRQn 1 */

  /* USER CODE END TIM6_DAC_IRQn 1 */
}
```

**How To Reproduce**
Create a project using an H725, enable FreeRTOS and set the timebase source to TIM6 and generate the project. Look at the ISR routine in stm32h7xx_it.c and you will see the ISR is missing the call to HAL_TIM_IRQHandler.

TOUNSTM, 2024-01-12:
Hello @Makin-Things,

First we would like to thank you for your report.

We tried to reproduce the problem you described for STM32H725 using STM32CubeMX v6.10.0 with no success. Could you please check another time with the same version as me and tell me if the problem persists.

With regards,

Makin-Things, 2024-01-12:
@TOUNSTM thanks for your response. I have checked by creating projects with both CubeMX 6.10.0 and CubeIDE 1.14.0 and both created the interrupt handler correctly. I copied the IOC from my problem project into the new CubeIDE project and when it generated the interrupt handler was broken.

I then did a comparison of the 2 IOC files and found a difference in this line.

Not working
```NVIC.TIM6_DAC_IRQn=true\:15\:0\:false\:false\:true\:false\:false\:false\:true```

Working
```NVIC.TIM6_DAC_IRQn=true\:15\:0\:false\:false\:true\:false\:false\:true\:true```

I changed this line with a text editor to match and then it generates correctly. I have no idea what this parameter means or how it got set differently considering that you can't edit anything to do with TIM6 once it is selected as the timebase.

TOUNSTM, 2024-01-17:
Hello @Makin-Things,

Let me tell you that these lines depend on the configuration you have set for `Time base: TIM6 `. In the `NVIC`, you'll find all the parameters relating to the `Time base`. You can therefore modify them. In your case, you need to check this checkbox (outlined  in RED in the Screenshot below).

I hope my comment is helpful. Please allow me to close this issue now.

With regards,

[image]

closed 2024-01-17 by TOUNSTM