STMicroelectronics/STM32CubeH7 #284: Wrong implementation of HAL_RCC_GetSysClockFreq, LL_RCC_CalcPLLClockFreq as well as SystemCoreClockUpdate in the examples

cryptic-quasar, 2024-03-01:
**Setup**
- Board: Nucleo H743ZI

**Issue**
I encountered a problem setting the UART baud rate using `LL_RCC_GetUSARTClockFreq(LL_RCC_USART16_CLKSOURCE)` as the second argument of `LL_USART_SetBaudRate`. The data received on my PC using the USB-to-Serial adapter were bogus. 

After spending hours, I finally spotted the issue and it turns out that the `SystemCoreClock` calculations in HAL and LL are wrong. The `PLLM` coefficient can only take values from 0 to 63 with 0 meaning division by 1, 1 meaning division by 2, 2 meaning division by 3, etc. The problem is that the methods I mentioned before do not take the increment of `PLLM` into account for the dynamic calculation of SystemCoreClock.

The following is taken from `stm32h7xx_hal_rcc.c`:
```
pllm = ((RCC->PLLCKSELR & RCC_PLLCKSELR_DIVM1) >> 4)  ;
```
where it must be:
```
pllm = ((RCC->PLLCKSELR & RCC_PLLCKSELR_DIVM1) >> 4) + 1;
```

TOUNSTM, 2024-03-08:
Hello @cryptic-quasar,

Thank you for this report. As indicated in the RM0433 (section 8.7.11 page 398) see screenshot below the prescaler for PLLM can take values from 0 to 63 this is correct but the value 0 indicates that the prescaler is disabled. I don't see any need to add the +1 in the PLLM calculation.
For the moment I didn't see the problem, may be I forgot some details. Could you please share with me more details about your issue.

With regards,
Tasnim

[image]

cryptic-quasar, 2024-03-08:
Hello @TOUNSTM,

Thanks for your reply. The setting and getting of PLLM are perfectly fine using `LL_RCC_PLL1_SetM()` and `LL_RCC_PLL1_GetM()`. But to use the result of `LL_RCC_PLLx_GetM()` for any frequency calculation, you must increment it by 1, otherwise, the result does not make any sense.
The problem lies in the following functions:
- `HAL_RCC_GetSysClockFreq()`
- `LL_RCC_GetPLL1ClockFreq()`, `LL_RCC_GetPLL2ClockFreq()` and `LL_RCC_GetPLL3ClockFreq()`
- `SystemCoreClockUpdate()`

These methods try to calculate the output clock frequency of a given PLL. Let's open up a small chunk of `LL_RCC_GetPLL1ClockFreq()` (parts of the code are redacted for visibility):

```
void LL_RCC_GetPLL1ClockFreq(LL_PLL_ClocksTypeDef *PLL_Clocks)
.
.
m = LL_RCC_PLL1_GetM();
.
.
if (m != 0U)
  {
    if (LL_RCC_PLL1P_IsEnabled() != 0U)
    {
      PLL_Clocks->PLL_P_Frequency = LL_RCC_CalcPLLClockFreq(pllinputfreq, m, n, fracn, LL_RCC_PLL1_GetP());
    }
.
.
```

Imagine `m` is set to 0 which is a perfectly valid variable that means division by 1. This method will not process it although it is valid. This is a bug on its own.

Now let's imagine `m` is set to 1, which means division by 2. Now, let's open up `LL_RCC_CalcPLLClockFreq()`:
```
uint32_t LL_RCC_CalcPLLClockFreq(uint32_t PLLInputFreq, uint32_t M, uint32_t N, uint32_t FRACN, uint32_t PQR)
{
  float_t freq;

  freq = ((float_t)PLLInputFreq / (float_t)M) * ((float_t)N + ((float_t)FRACN / (float_t)0x2000));

  freq = freq / (float_t)PQR;

  return (uint32_t)freq;
}
```
This division by 2 is not reflected anywhere. I would expect to see:
```
  freq = ((float_t)PLLInputFreq / (float_t)(M + 1U)) * ((float_t)N + ((float_t)FRACN / (float_t)0x2000));
```

Please let me know if there is any confusion left.

closed 2024-04-02 by cryptic-quasar