STMicroelectronics/STM32CubeH7 #286: initialisation setting for USB_OTG_HS as a disable HOST fails to return from init when using FW_H7  v1.11.2. (not so with v1.11.1

AnthonyGBES, 2024-04-04:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus) 

**Describe the set-up**
 * Custom Board based on stm32H753XI)
 * Keil IDE uVision 5.39 ARMclang compiler 6.21 STM32CubeMX 6.10.0 STCube FW_H7 1.11.2

**Describe the bug**
Setting up the USB_OTG_HS as host only and as internal Full speed PHY, and setting up the middleware USB_HOST to disabled. The Main() code never returns from MX_USB_OTG_HS_HCD_Init();
This was not the case with FW_H7 1.11.1  

**How To Reproduce**
USB Clock set to 48Mhz via PLL1 
enabling the pins for USB_OTG_HS and Set the USB_OTG_HS as host only and as internal Full speed PHY, but leave disabled in the middlewares Test with v1.11.1 and it will continue after function call to MX_USB_OTG_HS_HCD_Init(); (the reason for doing this is we have a HOST port connection there but it is not currently used)

Repeat same setup for v1.11.2 and it won't return from the MX_USB_OTG_HS_HCD_Init(); function.  

**Additional context**
I beleive the problem is it returns HAL_TIMEOUT from the call USB_CoreReset(); in both varients. but the call to USB_CoreInit(USB_OTG_GlobalTypeDef *USBx, USB_OTG_CfgTypeDef cfg) then void the return in HAL_HCD_init() in v1.11.1 and thus it carrys on regardless.

in v1.11.2 it checks the return value and then returns HAL_ERROR and gets caught by the Error_Handler(); within the MX_USB_OTG_HS_HCD_Init();

I suspect the real reason is why the USB_coreReset() is timing out???

**Screenshots**
If applicable, add screenshots to help explain your problem.

AnthonyGBES, 2024-04-04:
On closer inspection it appears that the STM32CubeMX has not set the __USB_OTG_HS_CLK_ENABLE() in this setup. This maybe the real reason it timesout.

AnthonyGBES, 2024-04-04:
Further Investigation has found STM32CubeMX has not created the HAL_HCD_mspInit() or HAL_HCD_mspDeInit().

When manually coding these it has fixed the problem.

So this should be passed to the STM32CubeMX team to fix as I have noticed there are many posts from people all having related problems to the Timeout issue here

ALABSTM, 2024-04-08:
Hi @AnthonyGBES,

Thank you for all these helpful details. However, I generated a project using the same configuration as you described and noticed no issue. Both `HAL_HCD_MspInit()` and `HAL_HCD_MspDeInit()` have been generated in the `stm32h7xx_hal_msp.c` file as you can see below. I used version **6.10.0** of `CubeMX` and version **1.11.2** of the `STM32CubeH7` firmware.

Would you mind double-checking please? Otherwise, please let me know in case I missed somehting.

[image]

Below the configuration I used:
* USB OTG HS peripheral internal PHY FS `Host Only`,
* USB Host Middleware `Disabled`.

[image]

[image]

With regards,

AnthonyGBES, 2024-04-08:
Do you also have generate separate .c/.h files set? 

When this is the case the MX_XXX_Init() and HAL_XXX_MspInit() functions are usually created in these seperate .c/.h files not the stm32h7xx_hal_msp.c. In this USB setup case it creates usb_otg_hs.c which has the MX_USB_OTG_HS_HCD_Init() but not the HAL_XXX_MspInit()/HAL_HCD_MspDeInit() functions. 

All the other modules seem correct and have both sets of functions UART, QSPI, ,LTDC etc.

[image]

Here's the sum total of my stm32h7xx_hal_msp.c file and below that the /* USER CODE BEGIN Header */

 /* USER CODE BEGIN Header */
/**
  ******************************************************************************
  * @file         stm32h7xx_hal_msp.c
  * @brief        This file provides code for the MSP Initialization
  *               and de-Initialization codes.
  ******************************************************************************
  * @attention
  *
  * Copyright (c) 2024 STMicroelectronics.
  * All rights reserved.
  *
  * This software is licensed under terms that can be found in the LICENSE file
  * in the root directory of this software component.
  * If no LICENSE file comes with this software, it is provided AS-IS.
  *
  ******************************************************************************
  */
/* USER CODE END Header */

/* Includes ------------------------------------------------------------------*/
#include "main.h"

/* USER CODE BEGIN Includes */

/* USER CODE END Includes */

/* Private typedef -----------------------------------------------------------*/
/* USER CODE BEGIN TD */

/* USER CODE END TD */

/* Private define ------------------------------------------------------------*/
/* USER CODE BEGIN Define */

/* USER CODE END Define */

/* Private macro -------------------------------------------------------------*/
/* USER CODE BEGIN Macro */

/* USER CODE END Macro */

/* Private variables ---------------------------------------------------------*/
/* USER CODE BEGIN PV */

/* USER CODE END PV */

/* Private function prototypes -----------------------------------------------*/
/* USER CODE BEGIN PFP */

/* USER CODE END PFP */

/* External functions --------------------------------------------------------*/
/* USER CODE BEGIN ExternalFunctions */

/* USER CODE END ExternalFunctions */

/* USER CODE BEGIN 0 */

/* USER CODE END 0 */
/**
  * Initializes the Global MSP.
  */
void HAL_MspInit(void)
{
  /* USER CODE BEGIN MspInit 0 */

  /* USER CODE END MspInit 0 */

  __HAL_RCC_SYSCFG_CLK_ENABLE();

  HAL_NVIC_SetPriorityGrouping(NVIC_PRIORITYGROUP_3);

  /* System interrupt init*/
  /* SVCall_IRQn interrupt configuration */
  HAL_NVIC_SetPriority(SVCall_IRQn, 6, 0);
  /* PendSV_IRQn interrupt configuration */
  HAL_NVIC_SetPriority(PendSV_IRQn, 7, 0);

  /** Enable the VREF clock
  */
  __HAL_RCC_VREF_CLK_ENABLE();

  /** Disable the Internal Voltage Reference buffer
  */
  HAL_SYSCFG_DisableVREFBUF();

  /** Configure the internal voltage reference buffer high impedance mode
  */
  HAL_SYSCFG_VREFBUF_HighImpedanceConfig(SYSCFG_VREFBUF_HIGH_IMPEDANCE_ENABLE);

  /* USER CODE BEGIN MspInit 1 */

  /* USER CODE END MspInit 1 */
}

/* USER CODE BEGIN 1 */

/* USER CODE END 1 */

This is the generated usb_otg_hs.c i have
/**
  ******************************************************************************
  * @file    usb_otg_hs.c
  * @brief   This file provides code for the configuration
  *          of the USB_OTG_HS instances.
  ******************************************************************************
  * @attention
  *
  * Copyright (c) 2024 STMicroelectronics.
  * All rights reserved.
  *
  * This software is licensed under terms that can be found in the LICENSE file
  * in the root directory of this software component.
  * If no LICENSE file comes with this software, it is provided AS-IS.
  *
  ******************************************************************************
  */
/* USER CODE END Header */
/* Includes ------------------------------------------------------------------*/
#include "usb_otg_hs.h"

/* USER CODE BEGIN 0 */

/* USER CODE END 0 */

HCD_HandleTypeDef hhcd_USB_OTG_HS;

/* USB_OTG_HS init function */

void MX_USB_OTG_HS_HCD_Init(void)
{

  /* USER CODE BEGIN USB_OTG_HS_Init 0 */
  
  /* USER CODE END USB_OTG_HS_Init 0 */

  /* USER CODE BEGIN USB_OTG_HS_Init 1 */

  /* USER CODE END USB_OTG_HS_Init 1 */
  hhcd_USB_OTG_HS.Instance = USB_OTG_HS;
  hhcd_USB_OTG_HS.Init.Host_channels = 16;
  hhcd_USB_OTG_HS.Init.speed = HCD_SPEED_FULL;
  hhcd_USB_OTG_HS.Init.dma_enable = DISABLE;
  hhcd_USB_OTG_HS.Init.phy_itface = USB_OTG_EMBEDDED_PHY;
  hhcd_USB_OTG_HS.Init.Sof_enable = DISABLE;
  hhcd_USB_OTG_HS.Init.low_power_enable = DISABLE;
  hhcd_USB_OTG_HS.Init.use_external_vbus = DISABLE;
  if (HAL_HCD_Init(&hhcd_USB_OTG_HS) != HAL_OK)
  {
    Error_Handler();
  }
  /* USER CODE BEGIN USB_OTG_HS_Init 2 */

  /* USER CODE END USB_OTG_HS_Init 2 */

}

ALABSTM, 2024-04-29:
Hi @AnthonyGBES,

I regenerated my project after having ticket the option `Generate ... as pair of '.c/.h' ... ` and I still get both `HAL_HCD_MspInit()` and `HAL_HCD_MspDeInit()` functions, but this time within a dedicated file named `usb_otg.c`, as you can see below.

It could be that you have ticked some other option at some moment, which causes this unwanted behavior. May I suggest that you restart creating a new IOC file from scratch with the tool's default options? Please keep me informed.

[image]

With regards,

ALABSTM, 2024-05-02:
Hi @AnthonyGBES,

I hope my previous comment is useful and you can easily overcome the issue you are facing. As suggested, it could be that you have ticked some other option at some moment, which causes this unwanted behavior.

Please allow me to close this thread. Thank you for your comprehension.

With regards,

closed 2024-05-02 by ALABSTM

AnthonyGBES, 2024-05-02:
It’s a lot of work to redo a 240 pin chip every time ST32CubeMX get confused!!!

It’s not just the pins that need setting up again but the whole 3 months project work. As all the user code sections will need redoing.

I think this is something you should really be investigating further and trying to fix.

What use is a code generator if you have to keep rebuilding it every time it gets confused.

I’m guessing all your setting up is the USB and not a more complicated project.

Try setting it to a different set up of usb E.g. a host msc from the same project you’ve set up as host disabled working project and see if it changes the files. And then try host disabling back again.

Perhaps it getting it right then first time but not if you change a project host use.

ALABSTM, 2024-05-06:
Hi @AnthonyGBES,

Really sorry to read that. Best of luck to update your project. Just one question about this sentence you wrote:
> As all the user code sections will need redoing.

Could you be more precise please? I guess your user code is correctly placed between the different `/* USER CODE BEGIN ... */` and `/* USER CODE END ... */` sections so it is regenerated along with the rest of the project each time.

With regards,