STMicroelectronics/STM32CubeH7 #287: Unexpected Closed Socket Error in AT_ExecuteCommand in WFI_connect function after Update

bnina-ayoub, 2024-04-04:
**Set-up**
- **The board**: STM32H7B3
- **IDE**: STM32CubeIDE
- **IDE Version**: 1.14.1
- **Firmware Version**: V3J13M4
- **Cube Programmer Version**: v2.16.0

**Bug Description**
I'm encountering an issue with the `AT_ExecuteCommand` function in the STM32H7 driver. Specifically, the function returns `ES_WIFI_STATUS_UNEXPECTED_CLOSED_SOCKET` when it encounters the `AT_ERROR_STRING` in the received data. This unexpected closed socket error occurs even when the AT command sent does not result in an error. I suspect there might be an issue with how the function is handling the received data or interpreting the `AT_ERROR_STRING`.

**How To Reproduce**
1. The global behavior of my application project involves sending AT commands to a device and handling the responses. This is not my proper application but the Clock And Weather example provided by ST.
2. I suspect the issue lies in the `AT_ExecuteCommand` function in the STM32H7 driver.
3. The problem occurs when the function encounters the `AT_ERROR_STRING` in the received data.
4. The problem can be reproduced by manually reimplementing the wifi connect function extracted from the provided Clock And Weather demonstration and executing it. The application can scan available WiFi APs and sends AT commands successfully, but the last command generates this error.

**Additional context**
It's worth noting that this code was executed and tested successfully before all updates. However, after updating, it now returns this unexpected closed socket error. I have not yet identified a patch or correction.

**Screenshots**

ES_WIFI_Connect function:

[image]

AT_ExecuteCommand function:

[image]

bnina-ayoub, 2024-04-07:
Hi @RJMSTM,

Just checking in on the status of my issue. Any updates ?

RJMSTM, 2024-04-08:
Hello @bnina-ayoub ,

It seems there might be some confusion regarding updates for `es_wifi` component. If there have been no updates for this component in four years.
[image]
it's important to understand which updates you're referring to.

Otherwise, If I understand correctly, the issue lies with **Clock And Weather** demonstration. It may not be related to an update within the component `es_wifi`, but to something else.
 What version of the STM32CubeH7 are you using?

Regards,

bnina-ayoub, 2024-04-08:
@RJMSTM Thanks for the response. 

Indeed the issue isn't related to the Network Library but it happened after updating Cube tools (CubeProgrammer, IDE and STLink firmware), not the es_wifi component itself. Hardware appears functional as access point scanning works, but joining the network with the AT C0\r command fails. Could there be configuration changes or compatibility issues causing this? I'm using STM32H7B3I-DK.

closed 2024-05-06 by ALABSTM