STMicroelectronics/STM32CubeH7 #289: A bug in the HAL_I2C_IsDeviceReady function in version 1.11.3

engycz, 2024-05-10:
I have found a bug in the HAL_I2C_IsDeviceReady function in version 1.11.3 fec141c.

The problem is in the function [HAL_I2C_IsDeviceReady](https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/fec141ce999da655a48e1a15db83a72d564a1312/Src/stm32h7xx_hal_i2c.c#L3251) on [Line 3332](https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/fec141ce999da655a48e1a15db83a72d564a1312/Src/stm32h7xx_hal_i2c.c#L3332). There is a call to the [I2C_WaitOnFlagUntilTimeout](https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/fec141ce999da655a48e1a15db83a72d564a1312/Src/stm32h7xx_hal_i2c.c#L6947) function . There is a new test for [I2C_IsErrorOccurred](https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/fec141ce999da655a48e1a15db83a72d564a1312/Src/stm32h7xx_hal_i2c.c#L7143) and if so (the device is not ready and sent NACK), I2C_IsErrorOccurred waits for STOP and returns HAL_ERROR to I2C_WaitOnFlagUntilTimeout. I2C_WaitOnFlagUntilTimeout returns HAL_ERROR to HAL_I2C_IsDeviceReady and it immediately returns HAL_ERROR to the upcoming function without considering the number of Trials from the function parameter.

**How To Reproduce**
1.  Write a data block into the EEPROM using the HAL_I2C_Mem_Write function 
2. Call HAL_I2C_IsDeviceReady with the Trials parameter set to 1000
3. HAL_I2C_IsDeviceReady return value is HAL_ERROR, only one busy test is performed

Version 1.11.2 - multiple attempts to test if the device is ready (not busy)
[image]

[image]

Version 1.11.3 - only one attempt testing if the device is ready (not busy)
[image]

KRASTM, 2024-05-14:
Hello @engycz,

Thank you for the report.

ST Internal Reference: 181377

KRASTM, 2024-05-15:
ST Internal Reference: 181377

degermann, 2024-08-23:
The change that's checked in does not seem to make sense with respect to the new local variable "status".  "Status" is set in several places, but it is not returned and it does not seem to affect the logic.  The only use of "status" that I can see is in the comparison on line 3370, where it is compared to HAL_ERROR and reset to HAL_OK if the current number of trials is less than the maximum number of trials.  But this doesn't affect anything because "status" isn't used anywhere else.  Am I missing something?