STMicroelectronics/STM32CubeH7 #290: v1.11.2 changes to lan8742.c result in hardfaults

Makin-Things, 2024-06-06:
**Describe the set-up**
Custom Board
STM32CubeIde 1.15.1

**Describe the bug**
I have updated a project from FW 1.11.1 to FW 1.11.2 and now it hardfaults. If I revert the single file Drivers/BSP/Components/lan8742/lan8742.c the hardfault no longer occurs.
Also with the v1.11.2 version of the lan8742.c file if I set a breakpoint on line 124 and step through the function and hit continue form line 130 all is good, I suspect this is because of the delay introduced b stepping through the code.

From the 1.11.2 release notes:

> BSP Components
> BSP component lan8742
> Remove unnecessary delay.

I think there is an argument that this was the wrong thing to do and the delay is in fact necessary.

ASEHSTM, 2024-06-10:
Hello @Makin-Things,

Thank you for your report. Could you please specify the exact microcontroller used on your custom board?

With Regards,

Makin-Things, 2024-06-11:
Hi @ASEHSTM

The exact device being used is a STM32H725IGT6.

Cheers
Simon

ASEHSTM, 2024-06-21:
ST Internal Reference: 184744

ASEHSTM, 2024-07-04:
Hello @Makin-Things,

Any update regarding this thread?

With regards,

ASEHSTM, 2024-07-05:
Hello @Makin-Things,

Please allow me to close this thread as no activity. You may reopen it at any time if you have any details to share with us. Thank you for your comprehension.

With Regards,