STMicroelectronics/STM32CubeH7 #291: CANFD update FW_H7 V1.11.2

cstyx, 2024-06-14:
**Describe the set-up**
 * STM32 NUCLEO-H723 board
 * STM32 CubeIDE 1.15.1 Build: 21094_20240412_1041 (UTC)
 * H723 firmware FW_H7 V1.11.2 

**Describe the bug**
* FDCAN_RxHeaderTypeDef property data_length is always 0 when receiving FDCAN messages

**How To Reproduce**
 * set up board for FDCAN (no bit rate switching)
 * receive a CAN-message **without** any data data_length == 0 -> **OK**
 * receive a CAN-message **with** some data (e.g. 8Bytes) the property data_length == 0 -> **FAIL**
 * probably CAN driver

ALABSTM, 2024-06-19:
Hi @cstyx,

Thank you for this report too. please refer to my other [comment](https://github.com/STMicroelectronics/STM32CubeH7/issues/292#issuecomment-2179218006). The two issues may be related.

With regards,

cstyx, 2024-06-20:
Hi @ALABSTM

as mentioned the FDCAN_RxHeaderTypeDef **property** data_length as provided by HAL_FDCAN_GetRxMessage() is always 0.

Regards

KRASTM, 2024-06-28:
Hello @cstyx,

Could you please share more details, like screenshots of `FDCAN_RxHeaderTypeDef` when the problem occurs.
In my tests with your project, the data_length **property** consistently shows a value other than 0.

with regards

cstyx, 2024-07-01:
Hi @KRASTM 

I attached a screen shots of the H723 sender. First is the tx_header and p_can_data as given to function `HAL_FDCAN_AddMessageToTxFifoQ(p_hcan, &tx_header, p_can_data)` containing 4 Bytes of data as indicated by DLC = 4 

[image]

th second one is the same message on the CAN BUS but the DLC seems to be 0 and data is already missing

[image]

cstyx, 2024-07-01:
Hi @KRASTM

I think I found the error at file stm32h7xx_hal_fdcan.c in function 
`static void FDCAN_CopyMessageToRAM(FDCAN_HandleTypeDef *hfdcan, FDCAN_TxHeaderTypeDef *pTxHeader, uint8_t *pTxData, uint32_t BufferIndex)` 

at code line  

`for (ByteCounter = 0; ByteCounter < DLCtoBytes[pTxHeader->DataLength >> 16]; ByteCounter += 4U)`

[image]

For CAN classic and CANFD the DLC for 4 bytes is "4" but this line shifts it 16 bit to the right so it will be always 0

KRASTM, 2024-07-02:
Hello @cstyx,

I don't understand, at first, you mentioned that you used FW_H7 V1.11.2, and now you suspect an error in FW_H7 V1.11.1!

Since the actual version of the FW_H7 is V1.11.2, and there were some updates related to FDCAN, include that line of code. So, I recommend you, to use the last version.

Thank you for your contribution, please allow me to close this thread.
 
with regards,

closed 2024-07-04 by KRASTM